openshwprojects/OpenBK7231T_App #1384: Configuring names in the device

bilogic, 2024-10-09:
**Describe the bug**
[image]

After my openBK device connects to homeassistant, I get 1, 2 and 3. It would be great if I could change that on the device itself

**Firmware:**
- Version 1.17.740
- Chip/model: BK7231N

**To Reproduce**
Shows up by default

**Screenshots**
See above

**Additional context**
No

MaxineMuster, 2024-10-09:
I'm not sure if it's the same request, but maybe you take a look at this PR, if that might be a possible solution.
https://github.com/openshwprojects/OpenBK7231T_App/pull/1336

bilogic, 2024-10-09:
@MaxineMuster thanks!

Yes, basically giving it a label, but I can't be certain your labels will be recognized by HA automatically, a particular way to structure the MQTT payload is required.

My view is that since these devices are rarely replaced, or changes what they control, it makes more sense to store them on the device itself. This way, if HA ever goes out of fashion, whatever else that comes along can show meaningful names when reading from MQTT.

openshwprojects, 2024-10-09:
I think you just need to use current SetChannelLabel without any PRs and do Hass Discovery, and the names will be seen by HA...

bilogic, 2024-10-09:
[image]

@openshwprojects thanks, HA sees the text I gave `SetChannelLabel`, but not applying them to the switches

openshwprojects, 2024-10-09:
I think you just need to reboot... or remove and readd device via discovery (but ifirst remove)

bilogic, 2024-10-09:
thanks it worked, I had to reboot HA.

closed 2024-10-09 by bilogic