openshwprojects/OpenBK7231T_App #1385: BK7231T and OTA update - every time loss autoexec

DamianLisu, 2024-10-10:
I have the same two [thermometers ](https://www.elektroda.com/rtvforum/topic3819498.html) based on BK7231T.
Both have the same problem. After OTA update (Web Application -> OTA -> Select file -> Start OTA) I am losing my autoexec file. After each update I have to recreate them. List Filesystem it doesn't show anything.
Autoexec is not complicated:
```
startDriver TuyaMCU
tuyaMcu_setBaudRate 9600
tuyaMcu_defWiFiState 4
startDriver NTP
ntp_timeZoneOfs +2

linkTuyaMCUOutputToChannel 1 val 0
linkTuyaMCUOutputToChannel 2 val 1
linkTuyaMCUOutputToChannel 9 val 2

delay_s 1
//F -> C
SetChannel 2 1
```
I will add that I have several devices with BK7231N and the described problem with the update does not occur here. I tried to restore the RF partition but without success. 
I have no idea what's going on.

DeDaMrAzR, 2024-10-11:
Can you share what was the previous version that you've updated from and to what version you updated that to?

There was a breaking change on T modules at some point that will erase LFS if OTA is done form an older version.

DamianLisu, 2024-10-12:
Of course, basically, the version doesn't matter now, because performing an upgrade from a version to the same version causes this problem. I currently have a version of 1.17.743.

Wolfieeewolf, 2024-10-22:
This is a long-standing issue

#1305 and #1010