openshwprojects/OpenBK7231T_App #1388: MQTT reconnects, but entities remain disabled in HA

bilogic, 2024-10-12:
**Describe the bug**
After I reboot my HA, my openBK device reconnects to the MQTT, but the entities remain disabled

[image]

**Firmware:**
- Version [e.g. 1.14.129]
- Device? 
- Chip/model: BK7231T?
- Device config?
```
Built on Oct 5 2024 17:12:00 version 1.17.740
Online for 3 days, 4 hours, 31 minutes and 20 seconds
Device MAC: C4:82:E1:56:B8:3A
Short name: obkB00083, Chipset BK7231N
```

**To Reproduce**
Steps to reproduce the behavior:
1. Reboot HA
2. Go to openBK IP address
3. MQTT state is connected, but HA shows the entities as disabled

**Screenshots**
[image]

**Additional context**
No

bilogic, 2024-10-14:
[image]

it's happening again

bilogic, 2024-10-20:
I have flags 10 and 27 checkboxed, but the issue remains...

bilogic, 2024-10-21:
I really regret flashing this instead of using tuyalocal, have to manually reboot my devices everytime HA reboots

openshwprojects, 2024-10-25:
Do you have flag <html>
<body>

2 | [MQTT] Broadcast self state every N (def: 60) seconds (delay configurable by 'mqtt_broadcastInterval' and 'mqtt_broadcastItemsPerSec' commands)
-- | --

</body>
</html>?

bilogic, 2024-10-25:
I did not touch that, but it is default correct? 

Can it reset itself once MQTT is disconnected so that it happens immediately after re-connection.

bilogic, 2024-11-10:
I give up, moving to ESPHome for way better reliability.

closed 2024-11-10 by bilogic