openshwprojects/OpenBK7231T_App #1390: Error: Failure on multiple SendGet Functions

codekow, 2024-10-14:
**Describe the bug**

## Command

Device fails to send multiple `SendGet` commands

```
alias scared_of_the_dark backlog sendGet http://10.0.40.51/cm?cmnd=Power0%20Toggle; sendGet http://10.0.40.52/cm?cmnd=Power0%20Toggle

addEventHandler OnDblClick 6 scared_of_the_dark
```

## Error

```
Info:HTTP_CLIENT:HTTPClient_Async_SendGet for http://10.0.40.52/cm?cmnd=Power0%20Toggle, sizeof(httprequest_t) == 160!
Info:HTTP_CLIENT:Parse url http://10.0.40.51/cm?cmnd=Power0%20Toggle
Info:HTTP_CLIENT:Parse url http://10.0.40.52/cm?cmnd=Power0%20Toggle
Info:HTTP_CLIENT:host: '10.0.40.51', port: 80
Info:HTTP_CLIENT:host: '10.0.40.52', port: 80
Info:HTTP_CLIENT:HAL_TCP_Establish: created socket 2

Error:HTTP_CLIENT:getaddrinfo error
Error:HTTP_CLIENT:establish connection failed, error -1 
Error:HTTP_CLIENT:httpclient_connect is error,ret = -8
``` 

**Firmware:**
- Version Build: Built on Sep 28 2024 17:02:26 version 1.17.731
- Device: 
- Chip/model: BK7231T
- Device config:

````
{
  "vendor": "Tuya",
  "bDetailed": "0",
  "name": "Full Device Name Here",
  "model": "enter short model name here",
  "chip": "BK7231T",
  "board": "TODO",
  "flags": "1024",
  "keywords": [
    "TODO",
    "TODO",
    "TODO"
  ],
  "pins": {
    "6": "Btn;1",
    "8": "LED_n;2",
    "9": "LED_n;1",
    "24": "Rel;1",
    "26": "dInput;2"
  },
  "command": "",
  "image": "https://obrazki.elektroda.pl/YOUR_IMAGE.jpg",
  "wiki": "https://www.elektroda.com/rtvforum/topic_YOUR_TOPIC.html"
}
```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

openshwprojects, 2024-10-15:
So it fails when you try to send two GET at the same time? But it works with single GET?

codekow, 2024-10-15:
@openshwprojects 

I am having inconsistent results. 

Usually one `SendGet` eventually completes - although the first time it may be a little delayed. But, I can never get both to complete. Sometimes the first one works, sometimes the second, but never both.

openshwprojects, 2024-10-16:
Maybe try adding a delay between them? In script, we dont have delay yet in backlog.

codekow, 2024-10-17:
@openshwprojects 

It seems like the second command is unable to resolve a network address until the first command completes. 

I haven't looked at the code closely yet, but some function may be blocking in `src/httpclient/utils_net.c`