openshwprojects/OpenBK7231T_App #1391: Error:CMD:cmd SENSOR NOT found in logs

jkwim, 2024-10-17:
**Describe the bug**
Error:CMD:cmd SENSOR NOT found in logs.

**Firmware:**
- Version: OpenBK7231N_1.17.740
- Device: WHDZ03 Smart Plug with Energy Monitoring (CB2S BK7231N + BL0937)
- Chip/model: BK7231N
- Device config:
- 
I have following flags enabled:

Flag 2 - [MQTT] Broadcast self state every N (def: 60) seconds (delay configurable by 'mqtt_broadcastInterval' and 'mqtt_broadcastItemsPerSec' commands)
Flag 10 - [MQTT] Broadcast self state on MQTT connect
Flag 21 - [MQTT] Retain power channels (Relay channels, etc)
Flag 30 - [MQTT] Enable Tasmota TELE etc publishes (for ioBroker etc) 

Drivers are running:
3 drivers active (Charts,NTP,BL0937), total 37 

**autoexec.bat**
```
alias mode_wifi setPinRole 8 WifiLED_n
alias mode_relay setPinRole 8 LED_n
// at reboot, set WiFiLEd
mode_wifi
// then, setup handlers
addChangeHandler WiFiState == 4 mode_relay 
addChangeHandler WiFiState != 4 mode_wifi 

startDriver BL0937
startDriver NTP
waitFor NTPState 1
ntp_setServer 192.168.1.235
ntp_timeZoneOfs +05:30

startDriver charts
// chart with max 48 samples, 2 variables and 2 vertical axes
chart_create 48 2 2
// set variables along with their axes
chart_setVar 0 "Voltage" "axv"
chart_setVar 1 "Power" "axp"
// setup axes
// axis_index, name, flags, label
chart_setAxis 0 "axv" 0 "Voltage (V)"
chart_setAxis 1 "axp" 1 "Power (W)"
addRepeatingEvent 5 -1 chart_addNow $voltage $power
```

**To Reproduce**
```
Info:MQTT:Publishing val (256 bytes) to tele/openbk7231N/SENSOR retain=0

Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic tele/openbk7231N/SENSOR

Error:CMD:cmd SENSOR NOT found (args {"Time":"2024-10-14T12:36:51","ENERGY":{"Power": 0.000000,"ApparentPower": 39.396476,"ReactivePower": 39.396476,"Factor":0.000000,"Voltage":237.316864,"Current":0.166007,"ConsumptionTotal":818.177917,"Yesterday": 423.476654,"ConsumptionLastHour":0.000000}})
```

**Additional context**
I had the plugs on 1.17.243 but unused for a while. I could not recall seeing this message in the log. After the upgrade I noticed the error. 

Also after bootup similar error is seen for RESULT:
```
Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic stat/openbk7231N/RESULT
Error:CMD:cmd RESULT NOT found (args {"Time":"2024-10-16T11:24:57","Uptime":"0T00:00:14","UptimeSec":14,"Heap":25,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":23,"POWER":"ON","Wifi":{"AP":1,"SSId":"xxxxxxxx","BSSId":"xxxxxxxxx","Channel":11,"Mode":"11n","RSSI":46,"Signal":-77,"LinkCount":21,"Downtime":"0T06:13:34"}})

```

jkwim, 2024-10-17:
I have tried increasing the TeleInterval:
`TasTeleInterval 10 60
`

And the error message now comes at intervals which match the TeleInterval except that there is a spurious error message which pops up every 70seconds. I could not figure out where the 70seconds come in.
(note that I have trimmed the text to about 140 columns to reduce the clutter)

```
$ nc 192.168.1.250 9000 | awk ' /SENSOR/ {print(strftime(), $0)}'
Thu Oct 17 11:27:09 +0530 2024 Info:MQTT:Publishing val (255 bytes) to tele/openbk7231N/SENSOR retain=0
Thu Oct 17 11:27:09 +0530 2024 Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic tele/openbk7231N/SENSOR
Thu Oct 17 11:27:10 +0530 2024 Error:CMD:cmd SENSOR NOT found (args {"Time":"2024-10-17T11:26:58","ENERGY":{"Powe
Thu Oct 17 11:27:10 +0530 2024 Info:MQTT:Publishing val (255 bytes) to tele/openbk7231N/SENSOR retain=0
Thu Oct 17 11:27:10 +0530 2024 Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic tele/openbk7231N/SENSOR
Thu Oct 17 11:27:10 +0530 2024 Error:CMD:cmd SENSOR NOT found (args {"Time":"2024-10-17T11:27:08","ENERGY":{"Powe
Thu Oct 17 11:27:20 +0530 2024 Info:MQTT:Publishing val (255 bytes) to tele/openbk7231N/SENSOR retain=0
Thu Oct 17 11:27:20 +0530 2024 Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic tele/openbk7231N/SENSOR
Thu Oct 17 11:27:20 +0530 2024 Error:CMD:cmd SENSOR NOT found (args {"Time":"2024-10-17T11:27:18","ENERGY":{"Powe
Thu Oct 17 11:27:30 +0530 2024 Info:MQTT:Publishing val (255 bytes) to tele/openbk7231N/SENSOR retain=0
Thu Oct 17 11:27:30 +0530 2024 Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic tele/openbk7231N/SENSOR
Thu Oct 17 11:27:30 +0530 2024 Error:CMD:cmd SENSOR NOT found (args {"Time":"2024-10-17T11:27:28","ENERGY":{"Powe
Thu Oct 17 11:27:40 +0530 2024 Info:MQTT:Publishing val (255 bytes) to tele/openbk7231N/SENSOR retain=0
Thu Oct 17 11:27:40 +0530 2024 Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic tele/openbk7231N/SENSOR
Thu Oct 17 11:27:40 +0530 2024 Error:CMD:cmd SENSOR NOT found (args {"Time":"2024-10-17T11:27:38","ENERGY":{"Powe
Thu Oct 17 11:27:50 +0530 2024 Info:MQTT:Publishing val (255 bytes) to tele/openbk7231N/SENSOR retain=0
Thu Oct 17 11:27:50 +0530 2024 Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic tele/openbk7231N/SENSOR
Thu Oct 17 11:27:50 +0530 2024 Error:CMD:cmd SENSOR NOT found (args {"Time":"2024-10-17T11:27:48","ENERGY":{"Powe
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
Thu Oct 17 11:27:58 +0530 2024 Info:MQTT:Publishing val (255 bytes) to tele/openbk7231N/SENSOR retain=0
Thu Oct 17 11:27:58 +0530 2024 Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic tele/openbk7231N/SENSOR
Thu Oct 17 11:27:58 +0530 2024 Error:CMD:cmd SENSOR NOT found (args {"Time":"2024-10-17T11:27:56","ENERGY":{"Powe
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
Thu Oct 17 11:28:00 +0530 2024 Info:MQTT:Publishing val (255 bytes) to tele/openbk7231N/SENSOR retain=0
Thu Oct 17 11:28:00 +0530 2024 Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic tele/openbk7231N/SENSOR
Thu Oct 17 11:28:00 +0530 2024 Error:CMD:cmd SENSOR NOT found (args {"Time":"2024-10-17T11:27:58","ENERGY":{"Powe
Thu Oct 17 11:28:10 +0530 2024 Info:MQTT:Publishing val (255 bytes) to tele/openbk7231N/SENSOR retain=0
Thu Oct 17 11:28:10 +0530 2024 Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic tele/openbk7231N/SENSOR
Thu Oct 17 11:28:10 +0530 2024 Error:CMD:cmd SENSOR NOT found (args {"Time":"2024-10-17T11:28:08","ENERGY":{"Powe
Thu Oct 17 11:28:20 +0530 2024 Info:MQTT:Publishing val (255 bytes) to tele/openbk7231N/SENSOR retain=0
Thu Oct 17 11:28:20 +0530 2024 Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic tele/openbk7231N/SENSOR
Thu Oct 17 11:28:20 +0530 2024 Error:CMD:cmd SENSOR NOT found (args {"Time":"2024-10-17T11:28:18","ENERGY":{"Powe
Thu Oct 17 11:28:30 +0530 2024 Info:MQTT:Publishing val (255 bytes) to tele/openbk7231N/SENSOR retain=0
Thu Oct 17 11:28:30 +0530 2024 Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic tele/openbk7231N/SENSOR
Thu Oct 17 11:28:30 +0530 2024 Error:CMD:cmd SENSOR NOT found (args {"Time":"2024-10-17T11:28:28","ENERGY":{"Powe
Thu Oct 17 11:28:40 +0530 2024 Info:MQTT:Publishing val (255 bytes) to tele/openbk7231N/SENSOR retain=0
Thu Oct 17 11:28:40 +0530 2024 Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic tele/openbk7231N/SENSOR
Thu Oct 17 11:28:40 +0530 2024 Error:CMD:cmd SENSOR NOT found (args {"Time":"2024-10-17T11:28:38","ENERGY":{"Powe
Thu Oct 17 11:28:50 +0530 2024 Info:MQTT:Publishing val (255 bytes) to tele/openbk7231N/SENSOR retain=0
Thu Oct 17 11:28:50 +0530 2024 Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic tele/openbk7231N/SENSOR
Thu Oct 17 11:28:50 +0530 2024 Error:CMD:cmd SENSOR NOT found (args {"Time":"2024-10-17T11:28:48","ENERGY":{"Powe
Thu Oct 17 11:29:00 +0530 2024 Info:MQTT:Publishing val (255 bytes) to tele/openbk7231N/SENSOR retain=0
Thu Oct 17 11:29:00 +0530 2024 Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic tele/openbk7231N/SENSOR
Thu Oct 17 11:29:00 +0530 2024 Error:CMD:cmd SENSOR NOT found (args {"Time":"2024-10-17T11:28:58","ENERGY":{"Powe
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
Thu Oct 17 11:29:08 +0530 2024 Info:MQTT:Publishing val (255 bytes) to tele/openbk7231N/SENSOR retain=0
Thu Oct 17 11:29:08 +0530 2024 Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic tele/openbk7231N/SENSOR
Thu Oct 17 11:29:08 +0530 2024 Error:CMD:cmd SENSOR NOT found (args {"Time":"2024-10-17T11:29:06","ENERGY":{"Powe
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
[... cut ...]
Thu Oct 17 11:32:40 +0530 2024 Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic tele/openbk7231N/SENSOR
Thu Oct 17 11:32:40 +0530 2024 Error:CMD:cmd SENSOR NOT found (args {"Time":"2024-10-17T11:32:38","ENERGY":{"Powe
Thu Oct 17 11:32:50 +0530 2024 Info:MQTT:Publishing val (255 bytes) to tele/openbk7231N/SENSOR retain=0
Thu Oct 17 11:32:50 +0530 2024 Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic tele/openbk7231N/SENSOR
Thu Oct 17 11:32:50 +0530 2024 Error:CMD:cmd SENSOR NOT found (args {"Time":"2024-10-17T11:32:48","ENERGY":{"Powe
```

openshwprojects, 2024-10-20:
I see, I've pushed a fix to silence it,  can you check now?

jkwim, 2024-10-21:
Flashed Built on Oct 20 2024 14:44:11 version 1.17.754

The error for SENSOR is suppressed now.

```
$ nc 192.168.1.218 9000 | awk '/SENSOR/ {print(strftime(), $0)}'
Mon Oct 21 11:26:36 +0530 2024 Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic tele/openbk7231N/SENSOR
Mon Oct 21 11:26:36 +0530 2024 Info:MQTT:Publishing val (256 bytes) to tele/openbk7231N/SENSOR retain=0
Mon Oct 21 11:26:36 +0530 2024 Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic tele/openbk7231N/SENSOR
Mon Oct 21 11:26:36 +0530 2024 Info:MQTT:Publishing val (256 bytes) to tele/openbk7231N/SENSOR retain=0
Mon Oct 21 11:26:36 +0530 2024 Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic tele/openbk7231N/SENSOR
Mon Oct 21 11:26:37 +0530 2024 Info:MQTT:Publishing val (256 bytes) to tele/openbk7231N/SENSOR retain=0
Mon Oct 21 11:26:37 +0530 2024 Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic tele/openbk7231N/SENSOR
Mon Oct 21 11:26:39 +0530 2024 Info:MQTT:Publishing val (256 bytes) to tele/openbk7231N/SENSOR retain=0
Mon Oct 21 11:26:39 +0530 2024 Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic tele/openbk7231N/SENSOR
Mon Oct 21 11:26:42 +0530 2024 Info:MQTT:Publishing val (256 bytes) to tele/openbk7231N/SENSOR retain=0
```

However the error for RESULT still appeared:

```
Mon Oct 21 11:24:32 +0530 2024 Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic stat/openbk7231N/RESULT
Mon Oct 21 11:24:32 +0530 2024 Error:CMD:cmd RESULT NOT found (args {"Time":"2024-10-21T11:24:30","Uptime":"0T00:02:19","UptimeSec":139,"Heap":25,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":23,"POWER":"ON","Wifi":{"AP":1,"SSId":"xxxx","BSSId":"xxxx","Channel":11,"Mode":"11n","RSSI":52,"Signal":-74,"LinkCount":21,"Downtime":"0T06:13:34"}})
```

openshwprojects, 2024-10-21:
I see, thanks, I've added Result stub now, can you check?

jkwim, 2024-10-22:
Flashed 1.17.755 and now the error is gone.

```
$ nc 192.168.1.218 9000 | awk '/RESULT/ {print(strftime(), $0)}'
Tue Oct 22 17:00:30 +0530 2024 Info:MQTT:Publishing val (309 bytes) to stat/openbk7231N-NW8-218/RESULT retain=0
Tue Oct 22 17:00:30 +0530 2024 Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic stat/openbk7231N-NW8-218/RESULT

```

Thanks a lot for the quick fixes!