openshwprojects/OpenBK7231T_App #1415: MQTT Set message received but ignored

balucio, 2024-11-06:
**The MQTT set message is received by device but it seems ignored**
When i try to set the relays status of the device using
```
mosquitto_pub -h localhost -t "luce-balcone/39/set" -u "username" -P "password" -m "1"
```
the message is received by device
```
Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic luce-balcone/39/set
```
but the relay is not triggered. It works from device webpage

**Firmware:**
- Version: 1.17.765
- Device: Generic Tuya Mini switch 1 ch (DS-1311AN)
- Chip/model: LN882H 
- Device config
```
{
  "vendor": "Tuya",
  "bDetailed": "0",
  "name": "Mini Smart switch",
  "model": "DN-1311AN",
  "chip": "LN882H",
  "board": "LN882H",
  "flags": "1029",
  "keywords": [
    "Tuya",
    "Smart Switch",
    "DS-1311AN"
  ],
  "pins": {
    "0": "dInput;5;39",
    "24": "Rel;39"
  },
  "command": "",
  "image": "https://obrazki.elektroda.pl/YOUR_IMAGE.jpg",
  "wiki": "https://www.elektroda.com/rtvforum/topic_YOUR_TOPIC.html"
}
  ```
- Startup
```
addEventHandler OnChannelChange 5 toggleChannel 39
setChannel 39 0
```

**To Reproduce**
Steps to reproduce the behavior:
1. Issue the MQTT message 
2. Check MQTT message is received by device and no Relay is triggered

openshwprojects, 2024-11-06:
Very strange, are you really sure that you're sending payload correctly? Does sending 0 works? What about Tasmosta-style "POWER TOGGLE" command?
Can you try changing your channel index to 0?

What if you remove ChangeHandler, does it work then?

openshwprojects, 2024-11-13:
pushed fix, thanks

closed 2024-11-13 by balucio