openshwprojects/OpenBK7231T_App #1422: docker build fails (after including ESP ??)

MaxineMuster, 2024-11-09:
**Describe the bug**
cloning repository and then running a docker build fails.

First error:

```
Cloning into '/OpenBK7231T_App/sdk/esp-idf/components/protobuf-c/protobuf-c'...
Cloning into '/OpenBK7231T_App/sdk/esp-idf/components/spiffs/spiffs'...
Cloning into '/OpenBK7231T_App/sdk/esp-idf/components/unity/unity'...
Submodule path 'sdk/esp-idf/components/bootloader/subproject/components/micro-ecc/micro-ecc': checked out 'fe8ed90abc356347378eb25e2586e90df7be015a'
Submodule path 'sdk/esp-idf/components/bt/controller/lib_esp32': checked out '5c4a62c1d4577d1352d28708c790ba2b4f741842'
fatal: Needed a single revision
Unable to find current origin/master revision in submodule path 'components/bt/controller/lib_esp32c2/esp32c2-bt-lib'
Failed to recurse into submodule path 'sdk/esp-idf'
make[1]: *** [Makefile:34: submodules] Error 1
make[1]: Leaving directory '/OpenBK7231T_App'
make: *** [Makefile:137: OpenBK7231T] Error 2

****************************************
```

Please see complete log in attachment

**To Reproduce**
Steps to reproduce the behavior:
1. clone repository 
2. change to `OpenBK7231T_App/docker` dir
3. call build: `docker run -it -v "$(pwd)/..":/OpenBK7231T_App  openbk_build`

Full log:
[OBK_docker_fails.txt](https://github.com/user-attachments/files/17688813/OBK_docker_fails.txt)

NonPIayerCharacter, 2024-11-15:
@MaxineMuster Does this happen if you manually add any esp to target_platforms in docker/build_target_platforms.sh?

NonPIayerCharacter, 2024-11-16:
For me, build succeeded after i removed --remote from git update submodules in makefile

MaxineMuster, 2024-11-16:
Sorry for my late reply

Adding/Building for an ESP target doesn't help, neither

    `docker run -it --env TARGET_SDKS="OpenESP32S2"  -v "$(pwd)/..":/OpenBK7231T_App  openbk_build`

did work nor

    adding `"OpenESPS2"` to` docker/build_target_platforms.sh`

helped.

But:

    removing `--remote ` did the trick!

Thanks, this really helps a lot, I'm able to build on my own again and am no longer dependent on "online-builds" for every small change!!!!!!