openshwprojects/OpenBK7231T_App #1432: Boot loop when MQTT is not configured

DT-one, 2024-11-19:
**Describe the bug**
The device appears to boot loop when MQTT is not configured. This is evident on a device with an LCD time display that goes to 00:00 or the ntp offset time every 1 to 2 minutes, which is what I will call the 'event'. Logs reveal autoexec.bat running after each event. It appears that MQTT is changing channel prior to most events. Configuring the MQTT settings to connect to a broker resolves the issue.

But, it would be useful for a device like this to be able to connect to wifi and just display the time, without having MQTT configured. I imagine there are other situations where it would be usefull for a device to run without MQTT.

It appears that http: page become non-responsive during an event. 

**Firmware:**
- Version: discovered on 1.17.689, also present on 1.17.740
- Device: S09 Thermometer and Humidity Sensor with IR (and LCD clock)
- Chip/model: BK7231T

**To Reproduce**
Steps to reproduce the behaviour:
1. Flash device
2. Set IP address and attach to WiFi
3. Set autoexec.bat with required entries to initialise NTP and TuyaMCU
4. Reset device
5. Watch LCD screen, See error

**Screenshots/Additional context**
Screenshots and logs available in the following elektroda thread, starting at post #18: https://www.elektroda.com/rtvforum/viewtopic.php?p=21307605#21307605

DT-one, 2024-11-20:
Through help provided on the elektroda thread:
The Tuya MCU may be resetting the Beken chip when 'no connection to cloud' is reported. Setting tuyaMcu_defWiFiState 4 fixes the issue.
But, an improvement may be to report 'connected to could' when MQTT is not configured, as case commonly encountered whilst setting up a device.

openshwprojects, 2025-06-21:
Added I guess

[image]

closed 2025-06-21 by openshwprojects