openshwprojects/OpenBK7231T_App #1442: Device doesn't wake up from PinDeepSleep with time parameter.

Zahy3100, 2024-11-27:
**Describe the bug**
If device is put to PinDeepSleep with time for example 10, it doesn't wake up after time passes. Wake up after pin action such as switch toggle works as expected. This issue occurs even when PowerSave is disabled.

**Firmware:**
- Version 1.17.774
- Device DIY smart switch
- Chip/model: BK7231T

**To Reproduce**
Steps to reproduce the behavior:
1. Go to console and send command PinDeepSleep 10
2. Click on send
3. Refresh web UI (it doesn't load because of deep sleep)
4. Wait aprox. 10 sec.
5. Refresh page and check if device woken up

openshwprojects, 2024-11-27:
As far as I know, this feature may not be present on T platform yet. We need to do some porting from N to T SDK. We haven't seen any T devices with deep sleep, you may be the first one to encounter it.
Related: 
https://github.com/openshwprojects/OpenBK7231N/commit/dac49553174ec52e0ca8b872e5c1e1d82d72d4a0

Zahy3100, 2024-11-27:
Okay that's interesting. Today I tested DeepSleep and it doesn't work either witch suits what you responded well. I'm thinking if it's just software issue because deep sleep works and it's able to wake up based on pin activity. I faced similar issue on esp01 where wasn't trace between pin of RTC circuit and RST pin. I'm thinking that this could be source of the issue here as well. I was trying to find more informations about that but there is no much of documentation to be found really. Maybe you have some datasheet you would like to share?😄 Also for more context I realized I didn't share exact board I'm using, it's WB3S. Thank you for reply! This project is awesome man!