openshwprojects/OpenBK7231T_App #1456: access to device data in html pages and scripts

rew1ndustry, 2024-12-05:
this is not a bug, it is a request for documentation, or a missing feature.

in the web pages and scripts stored in flash on the device, how to read

long name
short name
mac address
IP address

for use in these contexts.

some usage examples:

i have an html page for controlling the device, and i would like to display the long and short names.

i want to write a discovery script that fires on startup, to report the mac and IP addresses in my cloud using sendGet.

openshwprojects, 2024-12-05:
Use REST API and Tasmota STATUS command:
https://www.elektroda.com/rtvforum/topic3971355.html

rew1ndustry, 2024-12-05:
thank you very much for your helpful suggestion, however my target browsers cannot process javascript.

some form of macro replacement for $constants in html pages could be very useful, and should not be difficult to code.

also there does not seem to be any $constant form of long name, short name, mac address or IP address, in scripts.

i presume i must fork, and write my own, to get these features?