openshwprojects/OpenBK7231T_App #1473: Importing a template containing a command entry truncates the command after the first word

ReinoutHeeck, 2024-12-21:
**Describe the bug**
Importing a template containing a command entry truncates the command after the first word

**Firmware:**
- Version: Built on Dec 17 2024 12:29:10 version 1.17.802
- Device: TYWB 4ch-RF 
- Chip/model: CB3S / BK7231N
- Device config:
{
  "vendor": "Tuya",
  "bDetailed": "0",
  "name": "Tuya Smart Switch TYWB 4ch-RF AC/DC 7-32V Configured as Door Opener",
  "model": "TYWB 4ch-RF",
  "chip": "BK7231N",
  "board": "DC 7-32V",
  "flags": "136316032",
  "keywords": ["Door Opener"],
  "pins": {
    "8": "Btn;1",
    "22": "WifiLED_n;0",
    "24": "Rel;1"
  },
  "command": "addChangeHandler Channel1 == 1 addRepeatingEvent 0.5 1 setChannel 1 0"
}

**To Reproduce**

1. Using the web app import and apply the above template
2. in the 'Change Startup Command Text' page (http://1.2.3.4/startup_command) note that the text only contains the first word "addChangehandler" instead of the full command specified in the template.

openshwprojects, 2024-12-25:
Nice find, it seems that quotation mark was missing. I've added it here:
https://github.com/OpenBekenIOT/webapp/commit/0523072a4bcae19b63edc1d5d05658701565ce4a
Can you check if it works now?

ReinoutHeeck, 2024-12-25:
> Can you check if it works now?

Yep, that seems to fix it :-)