openshwprojects/OpenBK7231T_App #1502: Calibrating power metering with non-integers results in NaN power meter reading

divadiow, 2025-01-07:
Not looked at this extensively but I have been able to replicate an issue a user reported here https://www.elektroda.com/rtvforum/topic4096322.html

[image]

apply a non-integer value to power and readings on main GUI will be littered with NaN eg

[image]
setting back to integer does not fix

giedriuslt, 2025-01-16:
I guess it has something to do with NaN value. You can actually run comands 
```
VoltageSet NaN
PowerSet NaN
```
And they will work

DO NOT try, unless you can reflash with wires, in my case it just bricked the device.

giedriuslt, 2025-01-16:
Try build from #1513 I think that should fix the issue

divadiow, 2025-01-16:
this is the one persistent NaN through various resets and clears on GA fw,

[image]

That NaN is still there with 1513_merge_7263f06183e7 BUT settings 60.6 etc in power metering calibration no longer breaks the others.

giedriuslt, 2025-01-16:
Resetting energy stats should clear that one.

divadiow, 2025-01-16:
it does indeed ;)

PR #1513 (2025-01-17): Fix calibration
You can set 0 or NAN as calibration constant currently, which should not be allowed, and in my case bricks the device.
Should be validated correctly. 
Should fix #1502
closed 2025-01-17 by openshwprojects