openshwprojects/OpenBK7231T_App #1509: Controlling LED strip with WB3L chip not possible

xTreame, 2025-01-13:
i'm not sure if this is the right place for this

Yesterday I flashed OpenBeken on my WB3L (https://docs.libretiny.eu/boards/wb3l/#usage) chip. The chip is in a Meko LED Driver (MKP006C0160BLTG)
An LED strip is connected to it (RGBWC)
I have tried several configurations in Openbeken, including the GPIO Doctor. But the LED strip does not respond. It hangs on the same color that was set with the old/original firmware. 
An important info! There is another chip installed on the back. This is probably for the 2.4GHz remote control.
Do you have any experience with this? Do I have to set the config in OpenBeken to control the other chip? I just don't know how.

Thank you!

xTreame, 2025-01-13:
[image]
[image]
[image]

xTreame, 2025-01-13:
I found this document for the other chip
http://wiki.telink-semi.cn/doc/ds/DS_TLSR8367-E_Datasheet%20for%20Telink%202.4GHz%20RF%20System-On-Chip%20Solution%20TLSR8367.pdf

xTreame, 2025-01-13:
Thanks for your answer. I dumped the firmware before. But the tool was not able to extract some config (see Screenshot)
[image]

I will attach the dump
[readResult_BK7231T_QIO_2025-12-1-18-32-00.zip](https://github.com/user-attachments/files/18396866/readResult_BK7231T_QIO_2025-12-1-18-32-00.zip)

ataa, 2025-01-13:
It's a TuyaMCU, I managed to download it's Schema from Tuya using ltchip tool, There are Chinese names in the mode names:

```json
{
    "tokenResponse": {
        "expire_time": 300,
        "region": "EU",
        "secret": "63xR",
        "token": "HQolHjv8"
    },
    "deviceApi": "http://a.tuyaeu.com/d.json",
    "activeRequest": {
        "token": "HQolHjv8",
        "productKey": "brvmbe8zlxo0r58x",
        "softVer": "1.1.80",
        "protocolVer": "2.2",
        "baselineVer": "40.00",
        "cadVer": "1.0.2",
        "cdVer": "1.0.0",
        "options": "{\"isFK\":false}",
        "t": 1736773681
    },
    "activeResponse": {
        "capability": 1,
        "devId": "bf995c6e1f62a3ad46y1gi",
        "dstIntervals": [],
        "localKey": "c~9SP06tz:M<k2)=",
        "resetFactory": false,
        "schema": [
            {
                "mode": "rw",
                "property": {
                    "type": "bool"
                },
                "id": 1,
                "type": "obj"
            },
            {
                "mode": "rw",
                "property": {
                    "range": [
                        "white",
                        "colour",
                        "scene",
                        "scene_1",
                        "scene_2",
                        "scene_3",
                        "scene_4"
                    ],
                    "type": "enum"
                },
                "id": 2,
                "type": "obj"
            },
            {
                "mode": "rw",
                "property": {
                    "min": 12,
                    "max": 255,
                    "scale": 0,
                    "step": 1,
                    "type": "value"
                },
                "id": 3,
                "type": "obj"
            },
            {
                "mode": "rw",
                "property": {
                    "min": 0,
[... 268 lines cut ...]
        "online": false,
        "owner_id": "219857651",
        "product_id": "brvmbe8zlxo0r58x",
        "sub": false,
        "time_zone": "+04:00",
        "uuid": "abc6f37189f2872f"
    },
    "updateResponse": [
        {
            "channel": 0,
            "control_type": 0,
            "current_version": "1.1.80",
            "dev_type": 0,
            "last_upgrade_time": 0,
            "timeout": 0,
            "type": 0,
            "type_desc": "Main Module",
            "upgrade_status": 0
        }
    ],
    "deleteResponse": true,
    "cachedAt": 1736773685193,
    "cacheKey": "brvmbe8zlxo0r58x"
}
```

ataa, 2025-01-13:
[image]

xTreame, 2025-01-13:
Thanks!

I asked ChatGPT to translate the chinese symbols

开关 - Switch (On/Off)
模式 - Mode
亮度值 - Brightness value
冷暖 - Cool/Warm (Color temperature)
彩光模式数据 - Color light mode data
情景模式数据 - Scene data
炫彩模式 1 - Flash scene 1 (or: Colorful mode 1)
炫彩模式 2 - Flash scene 2 (or: Colorful mode 2)
炫彩模式 3 - Flash scene 3 (or: Colorful mode 3)
炫彩模式 4 - Flash scene 4 (or: Colorful mode 4)

ataa, 2025-01-13:
Check this guide:
https://www.elektroda.com/rtvforum/topic4038151.html

xTreame, 2025-01-13:
For the first test i added the autoexec.bat with this configuration

startDriver TuyaMCU
setChannelType 1 toggle
linkTuyaMCUOutputToChannel 1 1 1

I'm now able to turn LED strip on/off

openshwprojects, 2025-01-13:
Probably this can help: https://www.elektroda.com/rtvforum/find.php?q=tuyamcu_setupLED

xTreame, 2025-01-14:
Thank you.

I don't understand how to define the purpose of the channel

I have tried this autoexec entry

startDriver TuyaMCU
tuyaMcu_defWiFiState 4
tuyaMCU_setupLED 24 0

I have now the nice brightness, rgb and colour temp options on webinterface. but they don't change the real light.

How can i tell this config that it needs to use channel 3 for brightness and channel 4 for colour temp ??