openshwprojects/OpenBK7231T_App #1510: OpenBK7231T secondsWithNoPing autoexec bat no reset counter after ping back

wchpikus, 2025-01-13:
**Describe the bug**
A clear and concise description of what the bug is.

**Firmware:**
- 1.18.20
- Chipset BK7231N mini smart switch
- Chip/model: BK7231T?

autoexec:
startDriver NTP
startDriver SSDP

// setup ping watchdog
PingHost 8.8.8.8
PingInterval 10

// wait for NoPingTime reaching 60 seconds
waitFor NoPingTime 60
echo Sorry, it seems that ping target is not replying for over 60 seconds, will OFF switch!
// delay just to get log out
echo OFF NOW!
delay_s 1
SetChannel 1 0 
delay_s 10
SetChannel 1 1

Log:
Info:GEN:sta:rssi=-64,ssid=Data wifi,bssid=3e:f7:4e:b4:7b:e3,channel=6,cipher_type:CCMP
Info:MQTT:mqtt_host empty, not starting mqtt
Info:MAIN:Time 61, idle 186376/s, free 73216, MQTT 0(4), bWifi 1, secondsWithNoPing -1, socks 3/38 
Info:MAIN:Time 62, idle 188671/s, free 73216, MQTT 0(4), bWifi 1, secondsWithNoPing -1, socks 3/38 
Info:MAIN:Time 63, idle 193738/s, free 73216, MQTT 0(4), bWifi 1, secondsWithNoPing -1, socks 3/38 
Info:MAIN:Time 64, idle 194696/s, free 73216, MQTT 0(4), bWifi 1, secondsWithNoPing -1, socks 3/38 
Info:MAIN:Time 65, idle 190594/s, free 73216, MQTT 0(4), bWifi 1, secondsWithNoPing -1, socks 3/38 
Info:MAIN:Time 66, idle 193340/s, free 73216, MQTT 0(4), bWifi 1, secondsWithNoPing -1, socks 3/38 
Info:MAIN:Time 67, idle 201340/s, free 73216, MQTT 0(4), bWifi 1, secondsWithNoPing -1, socks 3/38 
Info:MAIN:Time 68, idle 194369/s, free 73232, MQTT 0(4), bWifi 1, secondsWithNoPing -1, socks 3/38 
Info:MAIN:Time 69, idle 194118/s, free 73216, MQTT 0(4), bWifi 1, secondsWithNoPing -1, socks 3/38 
Info:MAIN:Time 70, idle 192016/s, free 73216, MQTT 0(4), bWifi 1, secondsWithNoPing -1, socks 3/38 
Info:GEN:dhcp=0 ip=192.168.28.156 gate=192.168.28.183 mask=255.255.255.0 mac=a8:80:55:a3:f8:84
Info:GEN:sta: 1, softap: 0, b/g/n
Info:GEN:sta:rssi=-64,ssid=Data wifi,bssid=3e:f7:4e:b4:7b:e3,channel=6,cipher_type:CCMP
Info:MAIN:Time 71, idle 185964/s, free 73016, MQTT 0(4), bWifi 1, secondsWithNoPing 1, socks 4/38 
Info:MAIN:Time 72, idle 193198/s, free 73024, MQTT 0(4), bWifi 1, secondsWithNoPing 2, socks 4/38 
Info:NTP:NTP_CheckForReceive: Error while receiving server's msg
Info:MAIN:Time 73, idle 191868/s, free 73040, MQTT 0(4), bWifi 1, secondsWithNoPing 3, socks 4/38 
Info:NTP:NTP_CheckForReceive: Error while receiving server's msg
Info:MAIN:Time 74, idle 190380/s, free 73040, MQTT 0(4), bWifi 1, secondsWithNoPing 4, socks 4/38 
Info:NTP:NTP_CheckForReceive: Error while receiving server's msg
ExtraDebug:HTTP:DRV_SSDP_Send_Notify: space: 298 msg:244
ExtraDebug:HTTP:DRV_SSDP_Send_Notify: 
NOTIFY * HTTP/1.1
SERVER: OpenBk
HOST: 239.255.255.250:1900
CACHE-CONTROL: max-age=1800
LOCATION: http://192.168.28.156:80/ssdp.xml
NTS: ssdp:alive
NT: upnp:rootdevice
USN: uuid:1372d116-4163-4e8d-b6d0-f84a3d1df830::upnp:rootdevice

ExtraDebug:HTTP:DRV_SSDP_Send_Notify: sent message 244 bytes
Info:MAIN:Time 75, idle 190635/s, free 73040, MQTT 0(4), bWifi 1, secondsWithNoPing 5, socks 4/38 
ExtraDebug:HTTP:Received 244 bytes from 192.168.28.156
ExtraDebug:HTTP:data: NOTIFY * HTTP/1.1
SERVER: OpenBk
HOST: 239.255.255.250:1900
CACHE-CONTROL: max-age=1800
LOCATION: http://192.168.28.156:80/ssdp.xml
NTS: ssdp:alive
NT: upnp:rootdevice
[... 311 lines cut ...]
Info:GEN:dhcp=0 ip=192.168.28.156 gate=192.168.28.183 mask=255.255.255.0 mac=a8:80:55:a3:f8:84
Info:GEN:sta: 1, softap: 0, b/g/n
Info:GEN:sta:rssi=-76,ssid=Data wifi,bssid=3e:f7:4e:b4:7b:e3,channel=6,cipher_type:CCMP
Info:MAIN:Time 201, idle 184823/s, free 72864, MQTT 0(12), bWifi 1, secondsWithNoPing 131, socks 3/38 
Info:MAIN:Time 202, idle 191394/s, free 72864, MQTT 0(12), bWifi 1, secondsWithNoPing 132, socks 3/38 
Info:MAIN:Time 203, idle 190919/s, free 72864, MQTT 0(12), bWifi 1, secondsWithNoPing 133, socks 3/38 
Info:MAIN:Time 204, idle 192107/s, free 72864, MQTT 0(12), bWifi 1, secondsWithNoPing 134, socks 3/38 
Info:MQTT:mqtt_host empty, not starting mqtt
Info:MAIN:Time 205, idle 189753/s, free 72560, MQTT 0(13), bWifi 1, secondsWithNoPing 135, socks 3/38 
Info:MAIN:Time 206, idle 190459/s, free 72864, MQTT 0(13), bWifi 1, secondsWithNoPing 136, socks 3/38 
Info:MAIN:Time 207, idle 192543/s, free 72864, MQTT 0(13), bWifi 1, secondsWithNoPing 137, socks 3/38 
Info:MAIN:Time 208, idle 194475/s, free 72864, MQTT 0(13), bWifi 1, secondsWithNoPing 138, socks 3/38 
Info:MAIN:Time 209, idle 190921/s, free 60160, MQTT 0(13), bWifi 1, secondsWithNoPing 139, socks 4/38 
Info:MAIN:Time 210, idle 189115/s, free 72864, MQTT 0(13), bWifi 1, secondsWithNoPing 140, socks 3/38 
Info:GEN:dhcp=0 ip=192.168.28.156 gate=192.168.28.183 mask=255.255.255.0 mac=a8:80:55:a3:f8:84
Info:GEN:sta: 1, softap: 0, b/g/n
Info:GEN:sta:rssi=-80,ssid=Data wifi,bssid=3e:f7:4e:b4:7b:e3,channel=6,cipher_type:CCMP
Info:MAIN:Time 211, idle 190424/s, free 64240, MQTT 0(13), bWifi 1, secondsWithNoPing 141, socks 4/38 
Info:MAIN:Time 212, idle 201256/s, free 72864, MQTT 0(13), bWifi 1, secondsWithNoPing 142, socks 3/38 
Info:MAIN:Time 213, idle 188580/s, free 72648, MQTT 0(13), bWifi 1, secondsWithNoPing 143, socks 4/38 
Info:MAIN:Time 214, idle 192730/s, free 72416, MQTT 0(13), bWifi 1, secondsWithNoPing 144, socks 4/38 
Info:NTP:Seconds since Jan 1 1900 = 3945783003
Info:NTP:Unix time  : 1736794203
Info:NTP:Local Time: 2025-01-13 18:50:03
Info:MAIN:Time 215, idle 190254/s, free 72864, MQTT 0(13), bWifi 1, secondsWithNoPing 145, socks 3/38

openshwprojects, 2025-01-13:
well script does what it is instructed to, isn't it? If you want to reset and retry, try something like this, maybe:

startDriver NTP
startDriver SSDP

// setup ping watchdog
PingHost 8.8.8.8
PingInterval 10

again:

// wait for NoPingTime reaching 60 seconds
waitFor NoPingTime 60
echo Sorry, it seems that ping target is not replying for over 60 seconds, will OFF switch!
// delay just to get log out
echo OFF NOW!
delay_s 1
SetChannel 1 0
delay_s 10
SetChannel 1 1

ClearNoPingTime

delay_s 1

goto again

wchpikus, 2025-01-14:
Ok,i will try.. I was looking for goto again and clear no ping..
Tnx

divadiow, 2025-09-06:
if your chip is BK7231N.

closed 2026-09-18 by openshwprojects