openshwprojects/OpenBK7231T_App #1511: BL602 Daybetter MagicHome bulb not retaining settings

nospam2k, 2025-01-14:
**Describe the bug**
I have tried using flag 12 to retain settings but it doesn't seem to work.

**Firmware:**
- Version 1.18.20
- Device? Daybetter Magic Home bulb RGBWW
- Chip/model: BL602
- Device config? Defaults except for flag 12, 27, 30

**To Reproduce**
Steps to reproduce the behavior:
1. Set up bulb to wifi.
2. Check flags 12, 27, 30
3. Set bulb to WW, 100%, ON
4. Turn off bulb, turn on bulb, bulb is CW.
5. Also, set bulb to WW, 100%, OFF and do #4 and bulb is still CW and ON.

**Screenshots**
n/a

**Additional context**
n/a

openshwprojects, 2025-01-14:
https://github.com/openshwprojects/OpenBK7231T_App/blob/main/src/hal/bl602/hal_flashVars_bl602.c SaveLED is TODO, we can try to make it work if you have device with UART connect in case that smth goes wrong

nospam2k, 2025-01-14:
Unfortunately, I don't right now. I will work around it. I can use automations in HA.