openshwprojects/OpenBK7231T_App #1512: Frequenty mqtt disconnects with new versions

giedriuslt, 2025-01-15:
**Describe the bug**
New firwares disconnects from mqtt very often. It disconnects every hour or sometimes more frequently. Old version  from May was running solid with no disconnects. BL602 also seems to be running with no disconnects.

**Firmware:**
- Version 1.18.20
- Chip/model: BK7231N
- Config 
```
{
  "vendor": "Tuya",
  "bDetailed": "0",
  "name": "Full Device Name Here",
  "model": "enter short model name here",
  "chip": "BK7231N",
  "board": "TODO",
  "flags": "1028",
  "keywords": [
    "TODO",
    "TODO",
    "TODO"
  ],
  "pins": {
    "6": "BL0937CF1;1",
    "7": "BL0937CF;1",
    "8": "LED_n;1",
    "10": "Btn;1",
    "23": "WifiLED_n;1",
    "24": "BL0937SEL;1",
    "26": "Rel;1"
  },
  "command": "backlog startDriver BL0937; startDriver NTP; startDriver SSDP; ntp_timeZoneOfs 3:00",
  "image": "https://obrazki.elektroda.pl/YOUR_IMAGE.jpg",
  "wiki": "https://www.elektroda.com/rtvforum/topic_YOUR_TOPIC.html"
}
```
**Screenshots**
[image]
Old one was stable:
[image]

openshwprojects, 2025-01-19:
Are you that's all happening on the same device and is related to the firmware? Maybe capacitor degraded over time or something...

Can you narrow down which commit may introduce the problem?

Is it related to BL0937?

giedriuslt, 2025-01-19:
Capacitor would not cause disconnect, device is not rebooting itself. Also downgrading fixes the issue. 
I'm trying to find the version, there is more stuff published to mqtt lately, I think that might be causing this somehow.

giedriuslt, 2025-01-20:
Found the cause. Surprisingly fix for iobroker (version 17.670) is causing the issue while  (17.669) is stable. Very strange. Maybe my mqtt broker (mosquitto) does not support qos 2 and works like qos 0? Then qos 1 might mean waiting for acknowledgement from the broker and so filling the buffer, causing memory error and disconnect. Strange. Also strange that BL602 works without an issue 🤔.

[image]

openshwprojects, 2025-01-20:
Hmm I remember this qos stuff, it started with a strange error on ioBroker side and was being suprisingly fixed by setting qos... Well, it surely requires futher investigation, but we can check mqtt.c LWIP source code.  maybe we could put that to a flag?

If BL602 works well with this setting, then maybe we should compare the used LWIP libarries, or at least mqtt.c files to find out where is the difference?

giedriuslt, 2025-01-20:
Files are slightly different, so it's hard to pinpoint all changes, but change [one](https://github.com/openshwprojects/OpenBL602/commit/c73dcaab7e73ab68b29bd20a3b5d4c0992902c1d) is not present in bk7231. I'm also not sure which mqtt_opts.h is used for bk7231. One of them has small buffers...
And, I think we could send some stuff (for example all the voltage, current, power data) with qos 0, there is no need for them to have 1. That could free buffers faster.

giedriuslt, 2025-01-21:
I made a fix which works for me by sending energy messages with Qos zero. This makes it so that mqtt does not need to wait for acknowledgement from the broker and so has less buffers used. 
I think my bl602 might be working because it monitors device which is off most of the time, and as a result, it publishes much less energy messages.

PR #1518 (2025-01-22): Publish energy mqtt data with qos zero
This fixes mqtt disconnects for me. I thinkk publishing energy data with Qos 0 is appropriate because it's transient, and updates any few seconds anyway. Fixes #1512
closed 2025-01-22 by openshwprojects