openshwprojects/OpenBK7231T_App #1524: Issue with Data Transmission: Only Button Published, Data Not Sent to MQTT

Eddy001h, 2025-01-29:
**Issue with Data Transmission:**
When using the "Start Home Assistant Discovery" button, only toggle states are sent to MQTT in Home Assistant. However, numerical values such as temperature and similar data are not being transmitted.

- Version 1.18.21 - 1.18.30
- Chip/model: RTL87X0C
- Device config no Flags

--> **The settings under Configuration.**

[image]

--> **Under the Home menu, they are visible.**

[image]

[image]

-->**The button is being successfully published, but the data, such as sensor values, is not being transmitted to MQTT as expected.**

[image]

openshwprojects, 2025-01-29:
It seems that advanced discovey was simply  not enabled on RTL. I can enable it now, but I can't check if it works myself.
https://github.com/openshwprojects/OpenBK7231T_App/commit/8a3ecb5aafed87d81d1c74a394773454bc5642b1
Can you try latest release and let me know?

Eddy001h, 2025-01-29:
Perfect, now it’s working.
thx