openshwprojects/OpenBK7231T_App #1530: firmware update from 1.18.22 to 1.18.33 changes IP conf

wiltet, 2025-02-04:
**Describe the bug**
when update (in my case) from 1.18.22 to 1.18.33
--> "configure IP" all settings will be set to "192.0.0.0"
--> results in that the device will not be discoverable anymore
--> reboot to AP mode and set all to 0.0.0.0 to resolve the cause

**Firmware:**
- Version: 1.18.33
- Device:  Plug_Elivco_LSPA9_WL2S_LN882HKI
- Chip/model: LN882HKI/BK7231T

divadiow, 2025-02-04:
I have not been able to replicate this with a fresh LN 1.18.22 set to 0.0.0.0 OTA to 1.18.33

and neither with fixed ip 1.18.22 -> 1.18.33

wiltet, 2025-02-04:
Hm...
Not shure if it was set to "0.0.0.0" or "hu.hu.hu.hu" before. But I never did set something specific in "configure IP".
After the update I made today from 1.18.22 to 1.18.33 this happend. Same with a secound of these devices, but from a older build (1.17.xx).
Luckily there is still AP mode. Better than solder again. :)

openshwprojects, 2025-02-17:
Actually static IP was broken in older versions. The form was later fixed and static IP support was added. It's just that due to the previous bug you had bogus data in the form. It didn't break earlier because static IP was not implemented.

TLDR: I think your issue is a byproduct of a previous bug and a bugfix. There is nothing we can do now I guess..

closed 2025-02-18 by wiltet