openshwprojects/OpenBK7231T_App #1543: addChangeHandler Power < 20 appears broken

niteria, 2025-02-23:
Reading the documentantion I think I'm supposed to be able to do:

```
addChangeHandler Power < 20 echo low_power
```

I was never able to make it trigger, while:
```
addChangeHandler Power > 20 echo high_power
```

triggers.

I believe the line of code that's supposed to handle it is: https://github.com/openshwprojects/OpenBK7231T_App/blob/80111be27e34a3df00b532e6c64a6c08647e86fc/src/driver/drv_bl_shared.c#L693

It's guarded by some MQTT-related downsampling related to `VCPPublishIntervals` and `VCPPublishThreshold`, but even with `VCPPublishIntervals 0 60` and `VCPPublishThreshold 0.0001 0.0001 0.0001` I was still not able to trigger it.

Digging a bit more into the code I noticed that when I register the handler with `addChangeHandler` the `20` constant I pass is saved into `eventHandler_t`'s `requiredArgument2`:
* https://github.com/openshwprojects/OpenBK7231T_App/blob/80111be27e34a3df00b532e6c64a6c08647e86fc/src/cmnds/cmd_eventHandlers.c#L535
* https://github.com/openshwprojects/OpenBK7231T_App/blob/80111be27e34a3df00b532e6c64a6c08647e86fc/src/cmnds/cmd_eventHandlers.c#L325

But when the driver wants to check if it should trigger a change handler, `requiredArgument2` is never used:
* https://github.com/openshwprojects/OpenBK7231T_App/blob/80111be27e34a3df00b532e6c64a6c08647e86fc/src/driver/drv_bl_shared.c#L693
* https://github.com/openshwprojects/OpenBK7231T_App/blob/80111be27e34a3df00b532e6c64a6c08647e86fc/src/cmnds/cmd_eventHandlers.c#L299

To spell out the obvious, if it's unused I don't think it can work.

I'm a bit unsure if my conclusions are correct, because the MQTT-related downsampling logic makes it more complicated to tell what's happening and the conclusions are based on only reading the code.

What's a bit perplexing is that it works (produces output) one way, it seems that I'm more likely to end up with `sensors[i].lastSentValue > sensors[i].lastReading` than the other way around.

With a 40W lamp connected, typing `echo $power` into the console I'm able to get values of 0 and 40 depending on whether I turn my lamp on or off.

All of the code referenced is from the current HEAD of the `main` branch.

xjikka, 2025-03-23:
I checked it on the current version - 1.18.65 on BK7231N + BL0942, events seem to be fine.
Default settings, just set MQTT for logging values
autoexec.bat:

```
startDriver BL0942
addChangeHandler Power < 20 echo low_power
addChangeHandler Power > 20 echo high_power
```

logs:

```
Info:MQTT:Publishing val obk197087E8 to pwreventtest/host retain=0
Info:MQTT:Publishing val 0.124 to pwreventtest/current/get retain=0
Info:EVENT:EventHandlers_ProcessVariableChange_Integer: executing command echo high_power

Info:CMD:high_power
Info:MQTT:Publishing val 31.57 to pwreventtest/power/get retain=0

...

Info:MQTT:Publishing val 0.041 to pwreventtest/current/get retain=0
Info:EVENT:EventHandlers_ProcessVariableChange_Integer: executing command echo low_power
Info:CMD:low_power
Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic pwreventtest/current/get

...

Info:EVENT:EventHandlers_ProcessVariableChange_Integer: executing command echo high_power
Info:CMD:high_power
Info:MQTT:Publishing val 22.10 to pwreventtest/power/get retain=0

...

Info:EVENT:EventHandlers_ProcessVariableChange_Integer: executing command echo low_power
Info:CMD:low_power
Info:MQTT:Publishing val 0.00 to pwreventtest/power/get retain=0

...

Info:EVENT:EventHandlers_ProcessVariableChange_Integer: executing command echo high_power
Info:CMD:high_power
Info:MQTT:Publishing val 22.10 to pwreventtest/power/get retain=0

...

Info:EVENT:EventHandlers_ProcessVariableChange_Integer: executing command echo low_power
Info:CMD:low_power
Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic pwreventtest/current/get
Info:MQTT:Publishing val 15.78 to pwreventtest/power/get retain=0

...

```

Explanation:
`But when the driver wants to check if it should trigger a change handler, requiredArgument2 is never used:`
requiredArgument2  is translated to the eventType in CMD_AddChangeHandler using relationCode = EVENT_ParseRelation(relation);

```
static int EVENT_ParseRelation(const char *s) {
	if(!strcmp(s,"=="))
		return EVENT_TYPE_EQUALS;
	if(!strcmp(s,"!="))
		return EVENT_TYPE_NOT_EQUALS;
	if(!strcmp(s,">"))
		return EVENT_TYPE_GREATER;
	if(!strcmp(s,"<"))
		return EVENT_TYPE_LESS;
	if(!strcmp(s,">="))
		return EVENT_TYPE_EQUALS_OR_GREATER;
	if(!strcmp(s,"<="))
		return EVENT_TYPE_EQUALS_OR_LESS;
	return EVENT_DEFAULT;
}

```

Therefore is no longer needed in this mode,  eventType is checked using EVENT_EvaluateChangeCondition here:
```
	while(ev) {
		if(eventCode==ev->eventCode) {
			if(EVENT_EvaluateChangeCondition(ev->eventType, ev->requiredArgument, oldValue, newValue)) {
				ADDLOG_INFO(LOG_FEATURE_EVENT, "EventHandlers_ProcessVariableChange_Integer: executing command %s",ev->command);
				CMD_ExecuteCommand(ev->command, COMMAND_FLAG_SOURCE_SCRIPT);
			}
		}
		ev = ev->next;
	}

```

niteria, 2025-03-23:
Does it still work for you with MQTT disabled? I didn't have MQTT enabled

niteria, 2025-03-23:
I reread my analysis and I think this is the mistaken line:
> Digging a bit more into the code I noticed that when I register the handler with addChangeHandler the 20 constant I pass is saved into eventHandler_t's requiredArgument2:

I somehow misread the signature of `EventHandlers_AddEventHandler_Integer(byte eventCode, int type, int requiredArgument, int requiredArgument2, int requiredArgument3, const char *commandToRun)` and must have skipped over the `int type` argument mentally. The code I referenced calls it with `EventHandlers_AddEventHandler_Integer(eventCode,relationCode,reqArg,0,0,cmdToCall)`, so the 20 constant actually becomes `requiredArgument` and `requiredArgument2` is the default 0 and is irrelevant.

xjikka, 2025-03-23:
@niteria  - it is in the code by design checked only if MQTT enabled:

for checking the events then MQTT lastSentValue is used

`        EventHandlers_ProcessVariableChange_Integer(eventChangeCode, (int)sensdataset->sensors[i].lastSentValue, (int)sensdataset->sensors[i].lastReading);
`

```
#if ENABLE_MQTT
      if (MQTT_IsReady() == true)
      {
        sensdataset->sensors[i].lastSentValue = sensdataset->sensors[i].lastReading;

```

I have modified my PR https://github.com/openshwprojects/OpenBK7231T_App/pull/1578
and now it works even without MQTT.
Test it and let me know.
Wait a moment for the PR to build

xjikka, 2025-03-23:
https://github.com/openshwprojects/OpenBK7231T_App/actions/runs/14021330398/job/39253757956?pr=1578

niteria, 2025-03-23:
> https://github.com/openshwprojects/OpenBK7231T_App/actions/runs/14021330398/job/39253757956?pr=1578

This fixes it for me, thanks!

xjikka, 2025-03-23:
@openshwprojects  - modification included in https://github.com/openshwprojects/OpenBK7231T_App/pull/1578

closed 2025-03-26 by niteria