openshwprojects/OpenBK7231T_App #1546: Strange behavior of "flag" command

morganflint, 2025-02-24:
**Describe the bug**
After issuing the command "flags 0" (either from Web Application or from autoexec.bat), some flags are still set, even if they were manually cleared before

**Firmware:**
- Version 1.18.45
- Device: Atorch GR2P-WS 
- Chip/model: BK7231N

**To Reproduce**
Steps to reproduce the behavior:
1. Go to IP/cfg_generic
2. Clear all flags & press "Submit"
3. Go to Web Application -> Logs & type "flags 0"
4. Repeat 1 and see some flags activated

**Screenshots**
[image]

[image]

[image]

morganflint, 2025-02-24:
I made some additional tests: "Flags (Current value=X)" doesn't change setting or resetting flags higher than 31. 

For example:
With only flag 31 set it shows "Flags (Current value=1073741824)"
With only flag 31 set it shows "Flags (Current value=-2147483648)"
With only flag 32 set it shows "Flags (Current value=0)"
With flags 32 to 50 set, also 0

Issuing command "flags -2147483648" sets flags 0 to 30 to 0, and flag 31 to 1. In the range 32-50, flags 41, 42, 45, and 49 are set.
Issuing command "flags 2147483648" sets flags 0 to 30 to 1, and flag 31 to 0. In the range 32-50, flags 36, 38, 39, 42, 45, and 49 are set.
Issuing command "flags 1024" sets flag 10, and resets all the rest from 0 to 31. In the range 32-50, flags 37, 39, 40, 41, 45, and 49 are set.
Issuing command "flags -1024" resets flags 0 to 9 and sets flags 10 to 31. In the range 32-50, flags 37, 39, 40, 41, 45, and 49 are set.

So it seems the flags command accepts a 32-bit signed integer and sets flags 0 to 31 accordingly, but sets strange values to flags 32 to 50

openshwprojects, 2025-02-24:
Known bug, but we don't have a fix yet. I tried to attempt it, but it's still not good:
https://github.com/openshwprojects/OpenBK7231T_App/pull/1494/files

xjikka, 2025-02-26:
@morganflint  Can You check this PR: https://github.com/openshwprojects/OpenBK7231T_App/pull/1548

morganflint, 2025-02-28:
@xjikka 
I'm testing version 1548_merge_6d7a85401f09 on the same device. Some results:
"flags 0" effectively clears all the flags, but the text after "Current value =" shows a strange character:

[image]

"flags 2251799813685247" (2^51-1) sets all flags, and "Current value=2251799813685247" is shown correctly:

[image]

"flags 2251799813685248" (2^51) resets all flags and shows "Current value=2251799813685248":

[image]

Maybe it would be good to add a check to avoid "flags" argument higher than 2251799813685247, but probably nobody is going to try that :-)

"flags 2147483648" (2^31) sets flag 31 and resets all the rest. Shows the same in "current value"

"flags -2147483648" (-2^31) sets flags 31 to 50 and resets flags 0 to 30. Shows "Current value=18446744071562067968" (?)

"flags 2251797666201600" ( (2^51-1)-(2^31-1)) gets the same result, but shows "Current value=2251797666201600"

If I set some random flags and take note of the "Current value", then reset all of them (flags 0), and then give flags with the previous current value, the random flags are set again. Of course, the current value equals 2^a + 2^b + ... + 2^n, where a, b, ... n are the randomly set flags (this can be useful to save lines in the autoexec).

To summarize, the PR works well on my device if you don't try to force it with strange values. Maybe it'd be good to fix the strange character in "current value" when all flags are reset, and/or check valid values for the "flags" parameter, but these are not important issues.

xjikka, 2025-02-28:
`Thanks for fixing the bug!`
@morganflint  And thank You for the thorough and detailed test.

morganflint, 2025-03-01:
I've seen the PR is now merged, so I'll close the issue. Thanks!

closed 2025-03-01 by morganflint