openshwprojects/OpenBK7231T_App #1580: Feature request: auto-exit safe mode after predefined period of time

nalajcie, 2025-03-24:
Use case:
I have a remote site which has power failures from time to time. Some of the power failures make the smart plugs enter the safe mode (proably due to voltage fluctuations when the power grid is failing). The sockets stay indefinitely in safe mode until somebody manually disconnects them.

AFAIK it's not possible to disable safe mode or add some predefined reboot timeout in safe mode (autoexec is not being run, ping watchdog is disabled) without changing the code.

Please - add at least a hardcoded reboot timeout upon entering the the safe mode.

openshwprojects, 2025-03-27:
I think we can add this as a number of seconds stored in obkConfig_t structure, 0-255 range (0 means disabled). 

Or maybe just hardcode some value like 15 minutes?

nalajcie, 2025-03-30:
Probably adding it to `manConfig_t` would be most consistent as I see other timeouts are also stored there (https://github.com/openshwprojects/OpenBK7231T_App/blob/main/src/new_pins.h#L1367) - but this woud incur additional work with GUI, etc.

For my use case the hardcoded timeout is also OK.

I've checked the esphome and their "fallback AP" (in case you couldn't connect to any of the configured networks) doesn't have the auto-reboot feature out of the box (as a simple configurable parameter), but can be acheived with some YAML wizardy (eg. https://community.home-assistant.io/t/retry-wifi-connect-while-in-ap-mode/742764/1)