openshwprojects/OpenBK7231T_App #1595: MQTT connected always retained

psolyca, 2025-04-05:
**Describe the bug**
When the device connects to the MQTT broker a message is published to advice the connection.
But the message is send with the retain flag.

In Home Assistant, in my automation, I check for this connection to handle my firmware update but has the message is retained my automation fired automatically with a bad value.

Is it normal that this message should be retained ?

**Firmware:**
- Version from https://github.com/openshwprojects/OpenBK7231T_App/pull/1297
- Chip/model: BK7231N

**To Reproduce**
On each MQTT connection

openshwprojects, 2025-04-05:
First I'd check flags, maybe you have always retain flag on

psolyca, 2025-04-05:
No, I do not have the Flag 7, only 10, 19, 37 and 51 (from fast wifi PR)

openshwprojects, 2025-04-06:
Which exactly message has retain flag? "connected"?

giedriuslt, 2025-04-07:
Connected message should be retained, last will message makes it disconnected.

psolyca, 2025-04-07:
Sorry for the late reply, yes `OBK_DEV_NAME/connected` message is retained with the `online` value.

@giedriuslt, why should it be retained ?
If the device is not connected anymore for obvious reason, HA or other services will not see it and could try to communicate with it.
To be sure the device is still connected, a sort of heartbeat could be sent every N seconds !
BTW, in the https://github.com/openshwprojects/OpenBK7231T_App/blob/main/docs/mqttTopics.md, we have other mean to check the device is still alive, ` uptime`, `mac`... Maybe use the same principle with `connected`?

I could also use a variable to retain the connected status but when do I disable it ? In my particular case, low powered device, the wifi module is powered down by the MCU, the module does not have time to send any `offline` message, just the status of the device. So I need to check other "heartbeat" MQTT messages to be able to say my device is still alive. IMOO, it's too complicated.

psolyca, 2025-04-10:
But HA would also not know what is disconnected, retained message last "forever" !
Or I do not get the point here in technical terms.

If HA restart, HA will think that the device is connected.
But if it's not the case anymore, no more battery, connection troubles... HA has not the correct information.

How does devices other than OpenBK devices do ? I do not have the answer ^^;

giedriuslt, 2025-04-10:
MQTT has last will testament LWT message for that purpose. Then client disconnects broker itself will publish offline message for the device.

psolyca, 2025-04-10:
Hahahaha you could have explain it first :p
I did not know this ^^
I will have a look.
So I close this issue.

closed 2025-04-10 by psolyca