openshwprojects/OpenBK7231T_App #1598: How to save settings in custom driver?

Phil1pp, 2025-04-08:
I am trying to write a custom driver that requires to save some configuration data and restore it after reboot.
Is there already any function/library that can be used for this purpose?

Phil1pp, 2025-04-09:
Thanks for your answer, however 8 shorts won't be enough.
I am trying to implement a KNX interface which will need a storage of up to 1024 bytes.
Can I read and write to the Flash somehow?
Is LFS implemented an how to use it?

openshwprojects, 2025-04-09:
Yes we have LFS, check commands for lfs_write etc and check their
implementation. You can also check rest_interface.c file. You can read,
write, append and remove files. I can give more details later if it's needed