openshwprojects/OpenBK7231T_App #1612: LED Power Control Pin

mpevner, 2025-04-20:
Feit SHOP/4/HO/CCT/AG / BK7231N

The light Mostly works following configuration per https://github.com/openshwprojects/OpenBK7231T_App/issues/155 however I also have the additional bobbin I can't seem to figure out:

Pin 26 (PWM5) is the master on/off pin for the light control. If ignored, PWM2 (Brightness, channel 5) will bring the brightness down substantially at 0 but it is not completely off. Toggling PWM5 (treating as binary) will instead shut down the light completely with no remaining glow.

As it stands, On/Off will command PWM2 to 0 for off, but I can't seem to link PWM5 to this as well in a proper fashion (or just use it for on/off on its own).

openshwprojects, 2025-04-20:
can you provide more information, how many pwm pins are used for this light?

mpevner, 2025-04-20:
Sure!
Three pins in total, using Flag 8 for alternate CW light mode

these two seem to be configured correctly:
P7(PWM1) -> PWM_n, Color temperature, 4
P8(PWM2) -> PWM, Brightness, 5

The third pin I cant figure out the configuration for:
P26(PWM5) -> ?, Light on/off

mpevner, 2025-04-20:
To quickly add:
If I configure PWM5 as Relay, in the OFF state the light is on, and in the ON state the light is off.

Essentially a light off/on command needs to toggle the pin high/low instead of tweaking brightness, but I haven't figured out how to do that.

openshwprojects, 2025-04-20:
Use script, set target channel to led_enableAll every frame I guess ..  on
mobile rn.

mpevner, 2025-04-20:
Following that, I managed to cobble a basic setup for this:

P26(PWM5) has been configured as Rel_n channel 0

autoexec:
addChangeHandler Channel5 != 0 setChannel 0 1
addChangeHandler Channel5 == 0 setChannel 0 0

If there's a way to make the the Toggle 0 button go away, that'd be perfect but I think this can be closed otherwise.

closed 2025-04-20 by mpevner

openshwprojects, 2025-04-21:
maybe try to hide channel?

mpevner, 2025-04-21:
Oh! That worked perfectly, thank you!