openshwprojects/OpenBK7231T_App #1614: It stops communicating if it is not used.

IkaSGC, 2025-04-20:
Built on Jan 11 2025 18:38:56 version 1.18.19

Installed on the ceiling fan. Three main functions, light with dimming, fan on/off and speed reversal. 
All added via MQTT to home assistant. But if I don't use the fan for x hours, it stops responding. If I turn on the light for example, nothing happens and the entity state is unknown. I then have to go directly to MQTT and instead of the entity state I only see a lightning bolt symbols. Then I have to click several times "On/Off" to see the status switch again. 

It looks like the device has gone to sleep or stopped responding. Do you know what it could be ?

Setup:
Property: 
Light Switch: ID:20 - Bolean
Light WorkMode:  ID:21 - enum
Brightness: ID22 - Value (10-1000)
Light Temperature: ID23 - Value (0-1000)
Timer: ID:26 - Value (0-86400)

Fan Switch: ID60 - Bolean
Fan Speed: ID62 - Value 1-16
Fan Direction: ID63 - enum
Fan Countdown Left: ID:64 - unit 0-540min
Fan Beep: ID66 - Bolean

openshwprojects, 2025-04-21:
Does it reboot , or only HA loses state? Do you restart HA?
Do you have "broadcast self state every N minutes" or how it was called, flag?

IkaSGC, 2025-04-23:
Thanks for the response, this is my first time using this firmware, so I'm a bit confused. But as a result of your response, I checked the broadcasts in the settings and enabled them. Time will tell.

closed 2025-04-27 by IkaSGC