openshwprojects/OpenBK7231T_App #1637: Why isn't there a FULL scripting language reference available?

derei, 2025-05-16:
While there is a reference for console commands, I can't find any resources for full scripting syntax including event handlers and everything else.

For such a complex project are we supposed to navigate source code and pull syntax, or scour forums for script examples and hope the syntax isn't obsolete?

I understand devs and collaborators may be so familiar with it they don't need a reference - but I'm one of those users who can't read minds, and would rather read a documentation.

Thank you for reading - and hopefully for considering to publish a full syntax index.

openshwprojects, 2025-05-17:
The full scripting abilities are shown here:
https://github.com/openshwprojects/OpenBK7231T_App/blob/main/docs/autoexecExamples.md
But we also have now Berry support, which is explained here:
https://www.elektroda.com/rtvforum/topic4117238.html
Berry is recommended for more advanced stuff.

What are you trying to create, how may I help you?

RandomFunctionForUsername, 2026-02-13:
As a non-programmer berry's documentation isn't much better. But If you are willing to provide a script that would be awesome!