openshwprojects/OpenBK7231T_App #1639: Btn role does not emit OnClick events to scripting system in v1.18.100 (CB3S, BK7231N)

derei, 2025-05-16:
### Summary

On a iH001 RGBW LED Controller with **CB3S (BK7231N)** module running OpenBeken v1.18.100, button presses are detected at the hardware level (e.g. `key_short_press` appears in logs), but `OnClick` events are not emitted - neither to the console nor to registered `addEventHandler` scripts.

### Tested Configuration
- GPIO: P9
- Role: `Btn` (also tested `Btn_ScriptOnly`)
- Channels assigned: 10 (also tried 1, 11)
- PWM outputs on P7/P8/P24/P26 (for RGBW)
- Firmware: v1.18.100, flashed cleanly, dual power-cycle confirmed

### Observed Behavior
- Physical button press logs:
```
Info:GEN:9 Button_OnInitialPressDown
Info:GEN:9 key_short_press
```
- No `Event: OnClick X` appears
- No scripts fire even when handlers are registered via console:
```
addEventHandler OnClick 10 do log Hello_Click endon
```

### Confirmed
- `addEventHandler` confirms successful registration
- `autoexec.bat` correctly parsed on boot
- `logLevel 6` shows no event emission
- Tested all reasonable channel IDs (1, 10, 11)

### Additional Findings
autoexec.bat is successfully loaded and parsed (CMD_StartScript: started autoexec.bat at the beginning appears in logs)

However, the log command is not recognized (`cmd log NOT found`), meaning scripts silently fail at runtime if they contain log ...

This explains why registered handlers never produced output - the underlying scripting command set is incomplete or stripped down in the current firmware build

### Conclusion
Button press handling appears to be stuck in firmware-level detection, with **no dispatch to the scripting event system**. This breaks all automation using buttons unless the channel is auto-bound to hardware (e.g., relays).

Even if addEventHandler succeeds and button presses are detected at the GPIO level, no event logic can be verified or debugged due to the absence of the log command and failure of OnClick to fire.

### Ask
Please verify the channel event emission path from `Btn`/`Btn_ScriptOnly` roles into the `OnClick` dispatcher, or confirm whether this was fixed in a post-1.18.100 build.

Related: https://github.com/openshwprojects/OpenBK7231T_App/issues/1638

openshwprojects, 2025-05-17:
What is "do log" and "endon"?
Our syntax is here: 
https://github.com/openshwprojects/OpenBK7231T_App/blob/main/docs/autoexecExamples.md
Sample:

derei, 2025-05-17:
@openshwprojects I tried to run a script when pressing the button (an advancdd power on/off). But no matter what I tried, couldn't make it work.

The button press was detected as hw action, but won't trigger script execution even when event handler was correctly registered.

I'm either not understanding AT ALL what I'm doing (fair take, as it's my first time trying OBK), or there is a bug that interferes with my specific device.

derei, 2025-05-17:
@openshwprojects 
> What is "do log" and "endon"?

It's bad understanding of scripting on my part (lack of proper doc (I now believe the `do ... endon` is used only in SVM context).

openshwprojects, 2025-05-17:
Do endon looks like Tasmota syntax. We don't have it in Obk. I can consider
adding it , but I recommend berry for more advanced things

derei, 2025-05-17:
@openshwprojects best to keep things clean and not confusing - i'm confused already :)). So, if `do ... endon` is not part of obk, then perhaps keep it like that.

I will try berry.

btw, is the webapp down? I am getting only a blank page.

openshwprojects, 2025-05-17:
Hmm, Web app works only if you have internet connection, if you connect to
your device in AP mode and have no external connection, web app will not
load. Try pairing device with your WiFi first. Then it should work, I
guess...

Ok, maybe can you tell me what kind of script are you making? I can try
help with that

derei, 2025-05-17:
@openshwprojects I'm learning for now - abusing a £5 led strip controller, but actually I'm the one being abused.

I'm hitting the same walls with berry like I hit with OBK: same script, executed in different conrexts will work or not - and I feel I'm guessing blindly.
Actually not even guessing: so far, I was unable to make any working script (either autoexec.bat or by using berry) - commands work when sent in console or executed in svm, but when i call the script or expect it to execute as it was designed, it usually fails silently.

At this moment I'm wondering how you're all managing to work with such an unpredictable system.

openshwprojects, 2025-05-18:
I see how it can appear difficult. Would you like to try moving to some quicker messager like Google Chat or Discord so we can write your script together?

Maybe your old callbacks are kept or something.

openshwprojects, 2025-05-18:
Explanation: https://www.elektroda.com/rtvforum/topic4122454.html