openshwprojects/OpenBK7231T_App #1641: Calibrate or set threshold for touch sensitivity

alexDevBR, 2025-05-19:
Hi,
I have set the 6 flag to get immediate touch response, and it's great since I'm not using double click actions.
But I still feel like I have to press with a little more pressure than I had to before flashing. This is also mentioned in this issue: https://github.com/openshwprojects/OpenBK7231T_App/issues/1079

Is that something that's currently possible to adjust? If not, would it be hard to implement?

Thanks
Loving the project BTW, already convered all my wifi switches

DeDaMrAzR, 2025-05-21:
Hi @alexDevBR 

Touch sensitivity is not related to firmware it is being done (manufacturer depending) as a sort of a "digital capacitive latch" switch in the lack of a better term. So there is no way to calibrate that as it has nothing to do with the OBK controller/firmware. OBK is recognizing it as a simple switch.  

From my experience it is probably some sort of mechanical problem, maybe distance between the glass and "sensor" part changed in the disassembly. Or if it is a spring type maybe spring got compressed too much or something of that nature. 

Again without knowing the model, type and or the exact construction type these are only a general suggestions and advices you can try to improve the feel.

openshwprojects, 2025-06-18:
It's just not possible, touch controllers only give us 1 or 0. That's not ADC.

closed 2025-06-18 by openshwprojects