openshwprojects/OpenBK7231T_App #1654: LED control via IR TV remote blaster

C0rn3j, 2025-05-25:
[AliExpress listing](https://www.aliexpress.com/item/1005008769746264.html)

I bought a remote and some E14 LN882 Tuya bulbs.

The remote can bind up to 4 bulbs, by long-pressing the A/B/C/D buttons after turning off/on the bulb.

This works great with Tuya, but I cannot seem to set up a Tuya-flashed-Openbeken LN882 device this way with another E27 Tuya LN882 bulb (can't test the E14 one as I broke the one I was trying to flash).

I seem to be unable to find any references to this anywhere - what am I missing/how to set this up so the remote works for Openbeken, if that is even possible?

Remote in question, in case the listing is down:

[image]

DeDaMrAzR, 2025-05-26:
I suspect that remote is not IR based, maybe BT? If so then you are not going to achieve intended as OBK has not been delving into BT stack development at all (to my knowledge at least).

C0rn3j, 2025-05-26:
> I suspect that remote is not IR based, maybe BT?

Hmm I do not see any LED physically, nothing lights up on camera and covering the remote with my body makes it still work... so I suspect it is BT indeed.

> If so then you are not going to achieve intended as OBK has not been delving into BT stack development at all (to my knowledge at least).

That's unfortunate if so.

If someone picks it up, I am down to provide captures from the remote, if needed.

C0rn3j, 2025-07-04:
My spare has arrived.

To get into it, I pushed and pried with a shim on the bottom of the remote with minimal damage, there are two snap spots on the sides on the bottom of the remote, rest can be snapped by hand:

[image]

[image]

[image]

Markings identify the board/device as "tuya-remote-beacon-E-B1SD V1 xiaoma-office 2024/1/27".

There is a very clear and accessible UART port available.

The only chip of interest reads B1SD WE2I 2349ENF068 (initially misread it as WE21), this corresponds to B1SD-WE2I in [this Tuya doc](https://developer.tuya.com/en/docs/iot/chip-burn-and-authorization?id=Kceuttbn7mdc4) (in Chinese only)

[image]

[image]

[image]

[image]

It seems to look like I am ultra out of luck on having anything open running on this, there seems to be zero support I can find in the english-speaking internet sphere.