openshwprojects/OpenBK7231T_App #1669: 1.18.109 SHT3X Driver cant work, ma

sergiopoverony, 2025-06-02:
**Describe the bug**
A clear and concise description of what the bug is.

**Firmware:**
- Version [e.g. 1.18.109]
- Device? LightSwitcher
- Chip/model: BK7231N?
- Device config? 
```
{
  "vendor": "Tuya",
  "bDetailed": "0",
  "name": "Full Device Name Here",
  "model": "enter short model name here",
  "chip": "BK7231N",
  "board": "TODO",
  "flags": "33554436",
  "keywords": [
    "TODO",
    "TODO",
    "TODO"
  ],
  "pins": {
    "8": "LED_n;1;0",
    "10": "SHT3X_SCK;0;0",
    "11": "SHT3X_SDA;2;3",
    "23": "Btn;1",
    "24": "TglChanOnTgl;1",
    "26": "Rel;1"
  },
  "command": "",
  "image": "https://obrazki.elektroda.pl/YOUR_IMAGE.jpg",
  "wiki": "https://www.elektroda.com/rtvforum/topic_YOUR_TOPIC.html"
}
```

**To Reproduce**
Steps to reproduce the behavior:
I have 1.17.758 SHT work fine - after update to update 1.18.109 driver cant work. I cant see

openshwprojects, 2025-06-03:
Do you mean that driver is broken or that is just not included in build?

divadiow, 2025-06-05:
[OpenBK7231T_App_1670_merge_98514a7ea547_OpenBK7231N.zip](https://github.com/user-attachments/files/20615329/OpenBK7231T_App_1670_merge_98514a7ea547_OpenBK7231N.zip)