openshwprojects/OpenBK7231T_App #1671: LN822 has very poor wi-fi responsivity at times

C0rn3j, 2025-06-09:
**Describe the bug**
Most of the time, my two LN882 bulbs seem to be experiencing massive packet loss.

It does not seem to matter if the LED bulbs are actually off or on.

It seems that randomly toggling power on and on (physically) makes them sometimes go in a perfectly responsive state, but another reboot will be poor again.

**Firmware:**
- Version: 1.18.110 (second bulb is on a mildly older version with the same issue)
- Device: 
- Chip/model: LN882
- Device config: per device page is enough to reproduce

**To Reproduce**
Steps to reproduce the behavior:
1. Power the bulb on
2. Ping it

**Screenshots**

Issue when it's not that bad (it's at least eventually loading the web UI and configuring is possible, it can be worse): 

[image]

Perfect connectivity which is what I expect to always happen:

[image]

**Additional context**
Add any other context about the problem here.

gutek18a, 2025-06-11:
Try this version for example: https://github.com/openshwprojects/OpenBK7231T_App/releases/tag/1.17.650

I think there is a problem with newer firmwares, above >1.18.

divadiow, 2025-06-14:
Have you narrowed down the difference to a particular build?

divadiow, 2025-06-16:
that is quite far back but I have just tested latest OTA down to 1.17.650 and it worked. device was not configured with anything, no autoexec.

openshwprojects, 2025-06-20:
wait....  so which commit breaks stability? Currently I have no idea