openshwprojects/OpenBK7231T_App #1678: Add Wi-Fi Connection Watchdog Flag to auto-recover from disconnection

paulomac1000, 2025-06-18:
### 🚀 Feature Request: Add a Built-in Wi-Fi Connection Watchdog Flag

#### Summary
Please consider adding a simple built-in flag or configuration option (e.g., `wifi_auto_watchdog`) that enables automatic detection of Wi-Fi disconnection and performs a recovery action (Wi-Fi reconnect or full reboot) after a specified timeout.

---

#### Motivation
Devices running OpenBeken, located near the edge of Wi-Fi coverage, tend to silently drop off the network and **do not reconnect** unless manually rebooted or re-plugged.

In my case:
- The smart plug works fine initially, but eventually drops off the network.
- Sometimes it returns after a short while, but often stays disconnected until physically unplugged and replugged.
- A hardware antenna was added to improve reception, but it didn’t fully solve the issue.
- Plugging it back in restores normal function, which strongly suggests that the firmware **does not retry reconnection** or gets stuck in a failed Wi-Fi state.

This behavior is problematic in unattended or remote deployments.

---

#### Proposed Feature

Add a built-in mechanism like:

```text
setFlag wifi_auto_watchdog 1
setParam wifi_watchdog_timeout 300         # in seconds
setParam wifi_watchdog_action restartWiFi  # or "reboot"
#### Expected behavior:

-Periodically (e.g., every 60 seconds), check if the device is still connected (via getWiFiState() internally).
-If not connected for more than wifi_watchdog_timeout, perform the selected recovery action:
  - restartWiFi (non-intrusive),
  - reboot (full recovery if reconnect fails repeatedly)

This would offer a robust and reliable solution without requiring scripting or external automation.
#### Why This Matters

While it's currently possible to implement this logic via scripting or the PingWatchDog driver, these are not beginner-friendly or transparent solutions. A dedicated flag would:
- Make this behavior available out-of-the-box
- Improve resilience in poor Wi-Fi environments
- Reduce complexity and troubleshooting effort
- Help ensure devices don’t silently fall offline for hours or days

#### Optional Enhancement
Add a multi-stage recovery approach:
1. Attempt WiFiReconnect
2. If still offline after N attempts, then perform reboot

This would preserve uptime and avoid unnecessary reboots when a soft reconnect would have sufficed.
Thank you!

dummbold, 2025-07-31:
I totally agree! Restart on my smart lights take some minutes. I was trying to implement startup commands for:

- "Here I am" - red flash
- "Wi-Fi connected" - yellow flash
- "UI available" - green flash

red flash works, yellow flash is not possible - and don't ask me why the autoexec.bat (green flash) ist not executed.

So it would be great to have some watchdog flags during startup (especially for lamps) to make the progress of the startup visible.

schneemaier, 2025-11-19:
+1 for this feature