openshwprojects/OpenBK7231T_App #1683: Show associated Wi-Fi AP in `/index`

C0rn3j, 2025-06-22:
Sometimes I get `Wifi RSSI: Weak (-75dBm)`, sometimes it's excellent.

I would love to see the AP id, as I have multiple APs with the same SSID, and I wonder if such inconsistencies can be attributed to OpenBeken using the weaker AP, or any of the other things I am doing in between the changes at a glance, so having this info on this line would be great.

C0rn3j, 2025-06-24:
> but BSSID on main page was "too much" for other users

How about making it either visible on hover, or visible as an alt-text on hover?

That way it does not show unless you're actively hovering over the Wi-Fi line, not being overwhelming for regular users, but allowing power users to see it without digging into what seems to be a status JSON?

I can't unfortunately review the code itself, I don't do C.

MaxineMuster, 2025-06-24:
Exacly, hovering was my idea, too - see elektroda.com thread.
If you are willing to test, you could try by installing the "artifacts" from that PR on your device:

https://github.com/openshwprojects/OpenBK7231T_App/actions/runs/15853097645

You can then check "http://<IP.of.the.device>/cm?cmnd=STATUS%2011" (you can check it even now, but it will not show the actual BSSID but a fixed one)

C0rn3j, 2025-06-24:
Right, and the idea was to add it to the web-app instead for discoverability - are you going to send that in next? 

That'd be neat!

It'd be even more neat if it was on the main page as I can sometimes not even load the webapp properly on problematic devices, but knowing one can send a direct command for it is nice.

Frankly, I'd argue that the status page should have the BSSID in some form and that the MAC address and SSID should be ALSO hidden by default as sensitive information - I'll actually post more on that in the thread.

I currently don't have much time for testing but thank you for the PR.