openshwprojects/OpenBK7231T_App #1684: OpenBeken connects to a weak AP randomly when multiple SSIDs match the configured one

C0rn3j, 2025-06-22:
I have multiple APs with the same SSID.

One of them is downstairs, meters away from my two bulbs, one of them is upstairs, far too away from them.

Meanwhile one bulb connected to the good AP, one to the bad one, and as such is effectively inaccessible:

[image]

[image]

I would expect both to check which signal is better and connect to the one with better signal when detecting multiple APs, or at least connect to the stronger one eventually.

#1683 is mildly related to this issue

C0rn3j, 2025-06-24:
> It's a SDK problem, not OpenBeken.
> If a device is BK7231 or Realtek, this can be offset by enabling flag 51
> (will use bssid to connect).
> If LN882H, then that function would need to be implemented first.

I have devices on both BK and LN, I also have separate APs for 5 and 2.4GHz
networks that are per-device named so I can work around this by simply
throwing the network AP that makes the most sense, but I'd really prefer if
I could just not do that and use the unified AP.

So far I only spotted LN to be problematic for sure but it could be both.

If here is the wrong place to report it to, where should I do so to get it
fixed in the SDK?

MaxineMuster, 2025-06-24:
I also found this issue, but somehow fogott about...

IIRC the "better" AP was shown in the list, so if that is the case, I'm not sure, if changing "scan_time" can help. 
But that's some month ago...

If the APs are in fact all seen/scanned, I think, we can implement a workaround to enforce using the "best" AP.
I'm not at home, so sadly I can't test, but a basic idea it there:

inside wifi_scan_complete_cb all found SSIDs, BSSIDs and RSSI is reported.
So if we search for the lowest (I think it was lower is better?) RSSID of all APs sending the SSID we can set the BSSID of the AP.

I can try a PR for this, but can't test at the moment.