openshwprojects/OpenBK7231T_App #1701: Inability to do proper mutexing seems to be able to kill relays in some setups

C0rn3j, 2025-06-29:
I have electric roller blinds that share live and have a two input wires for connecting live to them to either go up and down.

They should not be able to trigger both states at the same time.

OBK only seems to allow to turn another input off after the current one already went on, resulting in a short time where both inputs are on - this seems to have fried two 10A230V relays of mine so far.

I would like the ability to do a proper mutex on two channels, ideally with a delay between the two.

Some more context in https://www.elektroda.com/rtvforum/viewtopic.php?p=21587087#21587087 and in https://github.com/openshwprojects/OpenBK7231T_App/issues/703#issuecomment-2980716857

Workaround for the time being would be welcome, I did not manage to come up with any.

openshwprojects, 2025-06-30:
Let's say channel 1 and 2 are mutexed and you do:
backlog setChannel 1 1; setChannel 2 1; setChannel 2 1; setChannel 2 0;
What should happen internally?

C0rn3j, 2025-06-30:
In my case, it definitely should not skip.

Imagine user presses blinds down, but it won't do anything, because they
already pressed blinds up accidentally x seconds earlier (the blinds take
half a minute to move completely).

So I would expect

```bash
# 1. Channel 1 ON
setChannel 1 1; 
# 2. Channel 1 OFF, delay 100ms, Channel 2 ON
setChannel 2 1; 
# 3. Nothing, channel 2 is already on
setChannel 2 1; 
# 4. Channel 2 OFF
setChannel 2 0;
```

openshwprojects, 2025-06-30:
preview sample: https://github.com/openshwprojects/OpenBK7231T_App/pull/1704 tested only in self test

openshwprojects, 2025-06-30:
Are you able to convert your script logic to use the proposed approach? A single channel with 3 states - 0 "do nothing", 1 "move down", and "2 move up" ?

C0rn3j, 2025-06-30:
I  will try flashing the artifact today on my second board, but how would I make this tri-state switch show up as "up" and "down" in home assistant?

openshwprojects, 2025-06-30:
i can add tristate channel type tomorrow, but for now, hm, a configuration.yaml custom entry?

openshwprojects, 2025-07-01:
ah it was not enabled on beken, i pushned new commit

openshwprojects, 2025-07-01:
ah yes, i moved pins to end, because i realized that the same approach could be used for N-pin mutexes

C0rn3j, 2025-07-01:
Okay I am failing here, mostly due to lack of understanding of channels/indexes.

```bash
setMutex 0 1 500 1 2
> Error:GEN:PinMutex[0] = ch=1, up=2, down=1, t=500ms
```

I don't get how to supply one channel when I am currently using two.

This is my current setup:

UP_1(1) / DOWN_1(2) / UP_2(3) / DOWN_2(4)
[image]

[image]

What am I doing wrong?

openshwprojects, 2025-07-02:
I used random pin indexes. This gen err is just a log message for quick testing.

I think you should not have set pins in html page. That can interfere with driver.

C0rn3j, 2025-07-02:
So so far I have this:

```bash
startDriver PinMutex
setMutex 0 0 500 6 24
setMutex 1 1 500 14 26

// This aliased command will turn off relay on Channel after 10 seconds
// addRepeatingEvent	[IntervalSeconds][RepeatsOr-1][CommandToRun]
alias turn_off_after_time_1 addRepeatingEvent 35 1 setChannel 0 0
alias turn_off_after_time_2 addRepeatingEvent 30 1 setChannel 0 0
alias turn_off_after_time_3 addRepeatingEvent 35 1 setChannel 1 0
alias turn_off_after_time_4 addRepeatingEvent 30 1 setChannel 1 0

// this will run the turn off command every time that Channel becomes 1 or 2
addChangeHandler Channel0 == 1 backlog turn_off_after_time_1
addChangeHandler Channel0 == 2 backlog turn_off_after_time_2
addChangeHandler Channel1 == 1 backlog turn_off_after_time_3
addChangeHandler Channel1 == 2 backlog turn_off_after_time_4
```

I have a couple issues with this setup (which otherwise now seems to work fine with the new driver):

---

1. I don't understand what this extra button column that's missing in the web app is:

[image]

[image]

2. I am not able to make the physical buttons work properly, I do not see how I can now make them toggle the separate UP/DOWN states.

3. The turn_off_after_time aliases do not die if the channel is set to 0, so if I toggle up, then down, there will be two timers running to turn the channel off at pretty much random times.

C0rn3j, 2025-07-02:
"Remember that some pin roles require second channel field, which is only available on native interface right now."

Right, that extra field business is on me, my bad.

Now I only have the issue with the turn_off aliases not dying if the state is changed more than once, which is not that bad.

I have ended up with this so far (I might have flipped up/down in the logic but that's besides the point, I'll fix that later):

```bash
startDriver PinMutex
setMutex 0 0 500 6 24
setMutex 1 1 500 14 26

// This aliased command will turn off relay on Channel after 10 seconds
// addRepeatingEvent  [IntervalSeconds][RepeatsOr-1][CommandToRun]
alias turn_off_after_time_1 addRepeatingEvent 35 1 setChannel 0 0
alias turn_off_after_time_2 addRepeatingEvent 30 1 setChannel 0 0
alias turn_off_after_time_3 addRepeatingEvent 35 1 setChannel 1 0
alias turn_off_after_time_4 addRepeatingEvent 30 1 setChannel 1 0

// this will run the turn off command every time that Channel becomes 1 or 2
addChangeHandler Channel0 == 1 backlog turn_off_after_time_1
addChangeHandler Channel0 == 2 backlog turn_off_after_time_2
addChangeHandler Channel1 == 1 backlog turn_off_after_time_3
addChangeHandler Channel1 == 2 backlog turn_off_after_time_4

// Aliases to be able to toggle between on/off state for each button state
alias click_up_1 if $CH0!=1 then "setChannel 0 1" else "setChannel 0 0"
alias click_down_1 if $CH0!=2 then "setChannel 0 2" else "setChannel 0 0"
alias click_up_2 if $CH1!=1 then "setChannel 1 1" else "setChannel 1 0"
alias click_down_2 if $CH1!=2 then "setChannel 1 2" else "setChannel 1 0"

addEventHandler OnClick 8 click_up_1
addEventHandler OnClick 9 click_down_1
addEventHandler OnClick 10 click_up_2
addEventHandler OnClick 11 click_down_2
```

C0rn3j, 2025-07-02:
> we can have it officialy, or just do a better shutter driver as a next step if you have time, would you be able to post later a summary and guide of our work to elektroda?

I don't mind, all the hard work on my side is done already anyways, though I suppose I'll wait til this is merged into the main build as to not to have to rewrite it later.

> maybe i could even do one step more and add addRepeatingEventUID (unique id) that automatically cancels previous instance?

That sounds neat and simpler for the user to script

openshwprojects, 2025-07-02:
added addRepeatingEventUID  , not tested yet

openshwprojects, 2025-07-06:
hey , any feedback, news?

C0rn3j, 2025-07-06:
I have not tried the repeating events, but I have been running the custom build (pinmutex_9c30a7621d99) without a hitch, so far the relays have survived, so I suppose lack of mutexing was indeed my root issue.

C0rn3j, 2025-08-05:
Anything I can do to help get this in master?

openshwprojects, 2025-08-10:
This can be easily merged, but probably not enabled by default due to flash size limitations

C0rn3j, 2025-08-12:
Does this mean it'd get a separate build, for example like IrRemoteESP does?