openshwprojects/OpenBK7231T_App #1709: Why getting log entry every second?

nospam2k, 2025-07-02:
**Describe the bug**
This is repeating every second:
Info:MAIN:Time 20, idle 0/s, free 112256, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/21 

**Firmware:**
- Version [latest]
- Device?  Light bulb
- Chip/model: BL602
- Device config? any

**To Reproduce**
Steps to reproduce the behavior:
1. Go to web application
2. Click on log
3. See error

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

DeDaMrAzR, 2025-07-03:
Because this is highly rapid developing environment an logs are something very useful if something goes wrong (which it can and has gone many times) 

If you don't want logs showing execute following command - loglevel 0 and enjoy the log silence 😉

nospam2k, 2025-07-03:
Ahhh... Ok, I thought maybe that was causing reboots. I am having random reboots and the light keeps turning on as default. (Apparently BL602 doesn't remember it's last state?). I can't use of for the default because it would have to be turned on. Glad for the info thx. I'll look into the reboot issue.

closed 2025-07-03 by nospam2k