openshwprojects/OpenBK7231T_App #1717: LN882h in smartplug model WHDZ01  Does not join my wifi

jn-safe4, 2025-07-08:
Firmware:
OpenLN882H_1.18.131
Smartplug model: WHDZ01
PCB markings: WH_EU_(N) V08 2024/07/31
Chip: LN882H

The plug was previously used with Cozylife.

I am able to upload the firmware successfully.
It starts the initial AP, and I can access it through 192.168.4.1.

I have tried connecting it to my home network with WPA2:
SSID: "Mew"
Password: "Password14"

However, I haven't been able to find any new devices on my network.

Using safemode to reconfig, I also tried connecting it to my phone's hotspot with no password:
SSID: "Mew3"

It does not connect there either.

From the UART console, on every power cycle, I get the following output:
"Console init ok!
ready"

To Reproduce
Flash the firmware (I have been using Itchiptool v4.12)
Configure Wi-Fi through the AP (192.168.4.1)
Power cycle the device
Observe that it does not connect to the Wi-Fi network
i can see from the power draw that it's in normal operation, not boot mode (100mA vs 24mA)

jonasnic, 2025-07-11:
Hmm.
Strange.
I think i was up to 5 reboots after config without ever seeing it on the network. maby i should just keept on.

I sort of gave up on that and have now 90% writen my own firmware. it was not to bad. exept for not beeing able to (know how) to do the ISR for the pulses. from the power chip.

just found out now while writing that the pins that the pulses are connected to; PB4 and PB5 dont support interrupts

jonasnic, 2025-07-11:
I see i have mixed my work and personal account.. but im am also jn-safe4

divadiow, 2025-07-18:
https://github.com/openshwprojects/OpenBK7231T_App/actions/runs/16367844373/job/46249571054?pr=1724

could give this a go?

MaxineMuster, 2025-08-02:
> From the UART console, on every power cycle, I get the following output: "Console init ok! ready"
> 
Probably to late, but which uart did you use?
LN882H has two: uart 0(A3/A2) and uart 1 (B8/B9), sometimes the naming is exchanged.
But it would be B8/B9 (in fact only Tx, so B9) to get the logs from OpenLN882H firmware