openshwprojects/OpenBK7231T_App #1748: Openbreken also have support for the RC6 OEM  Mode 6A?

displaydriver, 2025-08-04:
I have an Amino Aria Remote Control device but this does not work with Openbeken..

I have analyzed the IR protocol and see that the mode field 3 bits are: 110 (RC6 OEM Mode 6A)

[image]

divadiow, 2025-08-05:
which release of OBK are you running? default or irRemoteESP?

[image]

https://github.com/search?q=repo%3Aopenshwprojects%2FOpenBK7231T_App+path%3A%2F%5Esrc%5C%2Flibraries%5C%2F%2F+RC6&type=code

displaydriver, 2025-08-05:
Thanks for your replay!

I had installed the default OBK OpenBK7231N_QIO_1.18.145.bin, which was the problem. I've now installed OpenBK7231N_1.18.145_irRemoteESP.rbl via OTA and everything works great now. Thanks!