openshwprojects/OpenBK7231T_App #1752: Winner Micro W600 operate at temperature 86.2°C?

nw3000, 2025-08-07:
I have Winner Micro's W600 based devices flashed latest OpenW600_1.18.145, it works well! 

However I noticed in home page, looks like Chip temperature is around 86.2°C right after it booted up even without any other logic setup.
I also flashed another W600 device looks the Chip temperature is also close to 85°C even I just left the device running as a PCB board without outer shell.

is that a correct temperature information? because reading the datasheet  W600  maximum operation temperature is 85°C, i suspect 86.2°C is too high.

if that temperature correct , is it possible to lower the cpu running speed to lower the temperature?

MaxineMuster, 2025-08-09:
First: this is only a rough estimation about the chip temperature

Is this a lamp with W600?
They usually get quite hot because of the small housing and other components generating heat.

I only have a W800 dev board (so free air flow, no housing) which shows approx 57°C when idling .

Maybe @divadiow can help with some information on his W600 ?

MaxineMuster, 2025-08-09:
Will powersave work @divadiow ?
On my W800 it seems to have no impact (at least on the temperature).

divadiow, 2025-08-09:
TW-02/W600 module transplanted into mini breaker 

[image]
after a few mins online

divadiow, 2025-08-09:
powersave 1 on W600 drops draw from ~0.11A to ~0.2-0.5A

[image]

and now a drop in temp on Air602

[image]

divadiow, 2025-08-09:
TW-02 has dropped 20 degrees on powersave 1

MaxineMuster, 2025-08-11:
Just to clarify:  The "savetemps" firmware will not change anything but just add the graph to see the temperature over the time...