openshwprojects/OpenBK7231T_App #1755: WBRG1 - RTL8721 not responding

OlsMobile, 2025-08-09:
Hi guys,
I managed to flash OBK on a smart gateway, similar as shown in https://www.elektroda.com/rtvforum/topic4127578.html to replace the FW of https://github.com/MattWestb/EFR32-FW/issues/6 .

Flash with ImageTool went well and first AP connect worked fine : i added local network credentials, and reboot. IP is assigned and responding on the router, but i can't access : web conf timeout, nothing in HA or ImageTool, button push as no effect, any ideas ? Did i missed something ?

Thx

divadiow, 2025-08-09:
ordinarily 5 resets within reasonably quick succession should put it into safe mode and broadcast AP

OlsMobile, 2025-08-09:
Good to know, it works fine, many thanks !

divadiow, 2025-08-19:
oh wait. you could try emptying your autoexec.bat first and trying again in web app

NonPIayerCharacter, 2025-08-19:
Also turn on debug logging before ota. Either in web app or via command "loglevel 5"

NonPIayerCharacter, 2025-08-20:
Share log when you start OTA. That's what i meant. There can be some clues there.
Be sure to post log there guarded by triple ` symbols to keep formatting.

OlsMobile, 2025-08-20:
Tried on fresh reboot and get this :

```Debug:API:POST to api/ota
Debug:OTA:OTA post len 954400
Info:OTA:Current firmware index is 1
Error:OTA:Get OTA header failed
Error:OTA:OTA failed

closed 2025-08-20 by OlsMobile