openshwprojects/OpenBK7231T_App #1769: OpenBK7231N_1.18.156_irRemoteESP broken, everything works except for the blaster.

remlei, 2025-08-20:
**Describe the bug**
 OpenBK7231N_1.18.156_irRemoteESP broken, everything works except for the blaster.

**Firmware:**
- Version 1.18.156
- Device? Generic Tuya WIFI IR blaster
- Chip/model: BK7231N
- Device config
  "pins": 
    "7": "IRSend;1",
    "8": "IRRecv;1",
    "9": "Btn;0",
    "24": "WifiLED;1"

**To Reproduce**
Steps to reproduce the behavior:
Flash/Update to irRemoteESP variant.

**Screenshots**
The blaster part of the device is not working with irRemoteESP variant, flashing/ota u[date a  regular OpenBK7231N_1.18.156.rbl works just fine but with limited IR capabilities probably because its missing those IR stuff that is onto present on irRemoteESP variant. 

Besides that, the IR Receiver works. And was at least able to understand the IR on my split-type AC.

NonPIayerCharacter, 2025-08-20:
Try 1.18.153. If it works, then i've broken something in 1.18.154.

divadiow, 2025-08-26:
@remlei would you be able to give this a go? https://github.com/openshwprojects/OpenBK7231T_App/actions/runs/17114488224?pr=1771

remlei, 2025-08-26:
OpenBK7231N_1771_merge_c12381f18fdf_irRemoteESP.rbl

Flashed that firmware file and its still the same as before, no blaster, receiver works.

remlei, 2025-08-27:
OpenBK7231N_1775_merge_90d1ca590908_irRemoteESP.rbl

everything works now after flashing this firmware, both blaster and receiver works.