openshwprojects/OpenBK7231T_App #1785: Feature Request: Publish state when homeassistant/status = online

maxihatop, 2025-09-01:
**Problem:**
After Home Assistant restarts, the state of OpenBK devices is often shown as unknown, and device diagnostics are unavailable.
Currently, the only ways to restore state are:

enable periodic status broadcast (adds MQTT traffic and unnecessary flash wear),

manually reboot the OpenBK device,

toggle the device so it publishes something.

All three options are inconvenient.

**Proposed Solution:**
When the device receives the standard HA message

_homeassistant/status online_

it should immediately publish its full state (the same data normally published on broadcast interval).

**Benefits:**

Ensures HA quickly restores correct device state after restart.

Eliminates the need for periodic broadcasts (or allows them to be set to a much longer interval, e.g. 10 minutes).

Reduces MQTT traffic and flash wear.

Improves user experience: no manual toggling or rebooting required.

Summary:
→ Problem: after HA restart, OpenBK devices appear as unknown until they publish again.
→ Solution: publish full state once when HA announces itself online.
→ Result: correct state recovery, less traffic, more reliability.

openshwprojects, 2025-09-06:
Hm ok, can you check if it works ok?
https://github.com/openshwprojects/OpenBK7231T_App/pull/1791
PS: where do you get flash wear?

openshwprojects, 2025-09-07:
merged

maxihatop, 2025-09-07:
> PS: where do you get flash wear?
Sorry, there was my mistake, related to another possible issue. Irrelevant to this topic.

pnullpullbar, 2026-01-14:
Seems to be solved. Tested it today with 1.18.245 with 3 BK7231N-EU-Plug.