openshwprojects/OpenBK7231T_App #1812: [ESP8285]Core panic on mqtt set command with tuyaMCU

ahammer0, 2025-09-28:
**Describe the bug**
When the device recieves a MQTT set command that acts on tuyaMCU channel, it restarts ( due to core panicking ).

**Firmware:**
- Version 1.18.175
- Device noname-wifi dimmer, tuya compatible
- Chip/model: ESP8285 i flashed OpenESP8266 release
- Device config : no config change from the default except an autoexec.bat, Wifi and MQTT connected
```
// autoexec.bat
startDriver TuyaMCU

setChannelType 0 Toggle

//setChannelType 1 dimmer
//setChannelLabel 1 "Dimmer Level"

//setChannelType 2 TextField
//setChannelLabel 2 "Min dimming Level (10-1000)"

//setChannelType 3 TextField
//setChannelLabel 3 "Dimming mode (0,1,2)"

//tuyaMcu_setBaudRate 9600
//tuyaMcu_setDimmerRange 0 1000
linkTuyaMCUOutputToChannel 0 1 0
//linkTuyaMCUOutputToChannel 1 2 1
//linkTuyaMCUOutputToChannel 2 2 2
//linkTuyaMCUOutputToChannel 3 2 3
```
**logs from uart**
```
Info:MAIN:Time 329, idle 0/s, free 15744, MQTT 1(3), bWifi 1, secondsWithNoPing 263, socks 0/0 ␍␍␊
Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic obk_d_salon/0/set␍␍␊
Stack canary watchpoint triggered (quick)␍␊
␍␍␊
Backtrace: 0x4027f068:0x3fff4a50 Guru Meditation Error: Core  0 panic'ed (LoadStoreAlignment). Exception was unhandled.␍␍␊
Core 0 register dump:␍␊
PC      : 0x402441c3  PS      : 0x00000033  A0      : 0x4024413d  A1      : 0x3ffe9a10  ␍␍␊
A2      : 0x3fff4a6f  A3      : 0x000000c0  A4      : 0x00000022  A5      : 0x000000a0  ␍␍␊
A6      : 0x3ffe9a40  A7      : 0x4027d730  A8      : 0x3fff4a73  A9      : 0x000007fe  ␍␍␊
A10     : 0x00000090  A11     : 0x00000011  A12     : 0x4027d768  A13     : 0x3ffe9a50  ␍␍␊
A14     : 0x4027f068  A15     : 0x3fff4a50  SAR     : 0x0000001f  EXCCAUSE: 0x00000009  ␍␍␊
␍␍␊
Backtrace: 0x402441c3:0x3ffe9a10 0x40217b57:0x3ffe9a40 0x40100e84:0x3ffe9aa0 0x4024413d:0x3ffe9ac0 ␍␍␊
␍␊
 ets Jan  8 2013,rst cause:4, boot mode:(3,7)␍␊
␍␊
wdt reset␍␊
load 0x40100000, len 6244, room 16 ␍␊
tail 4␍␊
chksum 0x84␍␊
load 0x3ffe8008, len 24, room 4 ␍␊
tail 4␍␊
chksum 0x2c␍␊
load 0x3ffe8020, len 3296, room 4 ␍␊
tail 12␍␊
chksum 0x3d␍␊
```
**To Reproduce**
Steps to reproduce the behavior:
1. connect to MQTT broker
2. setup autoexec.bat as given
3. send mqtt set command

**Additional context**
I tried many different settings, if i disable tuyaMCU and comment lines "linkTuyaMCU..." i do not get the issue, the chip recieves the command ( i seen in uart log) but ignores them.

I tried to set the flag 26 to use alternate UART for tuyaMCU, thinking that it could be the tuyaMCU driver conflicting with the uart log: the tuyamcu frames were sent to the other uart (UART1) but still have the issue.

I tried the logport ( and logPort, in case the typo matters) to change the log to UART 1 but the command is not recognised.

ahammer0, 2025-09-29:
@NonPIayerCharacter Thank you for your quick anwser.

I did flash your fix.

Also got core crash when issuing HASS discovery from webapp.
```
Info:MAIN:Time 39, idle 0/s, free 14992, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 0/0 ␍␍␊
+--------------- net device info ------------+␍␍␊
|netif type    : STA                         |␍␍␊
|netif rssi    = -43                         |␍␍␊
Info:MAIN:Time 40, idle 0/s, free 14992, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 0/0 ␍␍␊
|netif ip      = 192.168.1.12                |␍␍␊
|netif mask    = 255.255.255.0               |␍␍␊
|netif gateway = 192.168.1.1               Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.␍␍␊
Core 0 register dump:␍␊
PC      : 0x4022ae84  PS      : 0x00000030  A0      : 0x4022ae7d  A1      : 0x3fffd7d0  ␍␍␊
A2      : 0x00000600  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x2c880300  ␍␍␊
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x6e007d22  A9      : 0x00000022  ␍␍␊
A10     : 0x0000005c  A11     : 0x00000002  A12     : 0x3fffec54  A13     : 0x3fffe343  ␍␍␊
A14     : 0x3fffd900  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001d  ␍␍␊
␍␍␊
Backtrace: 0x4022ae84:0x3fffd7d0 0x4022af2c:0x3fffd800 0x402241e4:0x3fffd810 0x4022442d:0x3fffd900 0x4022757d:0x3fffd930 0x4022507e:0x3fffd960 Guru Meditation Error: Core  0 panic'ed (LoadStoreAlignment). Exception was unhandled.␍␍␊
Core 0 register dump:␍␊
PC      : 0x4024429f  PS      : 0x00000033  A0      : 0x40244219  A1      : 0x3ffe9a10  ␍␍␊
A2      : 0x3fffd98d  A3      : 0x000000c0  A4      : 0x00000030  A5      : 0x000000a0  ␍␍␊
A6      : 0x3ffe9a40  A7      : 0x40224f98  A8      : 0x3fffd991  A9      : 0x000007fe  ␍␍␊
A10     : 0x00000090  A11     : 0x00000011  A12     : 0x4022ae7d  A13     : 0x3ffe9a50  ␍␍␊
A14     : 0x4022507e  A15     : 0x3fffd960  SAR     : 0x0000001f  EXCCAUSE: 0x00000009  ␍␍␊
␍␍␊
Backtrace: 0x4024429f:0x3ffe9a10 0x40217b77:0x3ffe9a40 0x40100e84:0x3ffe9aa0 0x40244219:0x3ffe9ac0 ␍␍␊

```

So i tried to deactivate tuyaMCU driver so the discovery passses, then reactivating the driver and trigger state change on the toggle channel.
It did work !

PS: Any ideas, why "logport" command is unknown ?
```
Error:CMD:cmd logport NOT found (args 1)
Info:CMD:[WebApp Cmd 'logport 1' Result] Unknown command
```

NonPIayerCharacter, 2025-09-29:
There was probably not enough free ram for discovery to work.

logport is BK72xx only.

PR #1814 (2025-09-29): Increase stack size for ESP8266
Fixes #1812?